markniu/Bed_Distance_sensor #80: Monster 8 connection

joejoe317, 2024-01-24:
My board should be here tomorrow so I am prepping by figuring out which gpio I can connect the bd sensor to my monster 8.

It is mentioned the normal BL touch connection will not work, and to use gpio. It looks like the few posts that mention bltouch with monster 8 are connecting to PA8 and any combination of z endstop. 

Am I trying to avoid both the z endstop pins and PA8, or just PA8?

is it okay to connect to EXP1 that has 5v ground and some other gpio PE7,PD8,PD10,PE10,PE15,PD9,PE11,PB2

markniu, 2024-01-24:
right, you can use all the pins in exp1, like the 5v,ground and any other pins in exp1 for the clk and sda.

joejoe317, 2024-01-24:
> right, you can use all the pins in exp1, like the 5v,ground and any other pins in exp1 for the clk and sda.

Thanks! Just for future documentation is it specifically pa8 that we are trying to avoid?

markniu, 2024-01-24:
the PA8 is OK, because it is connected to the mcu directly and no capacitor on that pin, we need to avoid the endstop pins.