markniu/Bed_Distance_sensor #81: More squish

joejoe317, 2024-01-30:
What settings do I change to give more squish on first layer? My boat keep breaking off the print bed and seems the first layer isn’t adjusted very well. 

Is there docs on what each bdsensor setting does in klipper?

szabypap, 2024-02-09:
Following! I have a trick. I start printing and baby step while printing when I am happy with the first layer I nite the height of stepping like 0.05. Stop printing, home and I move the nozzle to 0.05 and from there I give an other calibration

dhoard, 2024-04-29:
I'm running Klipper and found this procedure to be consistent and work as expected.

1. Set `z_adjust=0`
2. Perform calibration
3. Print a first layer test, baby stepping Z offset to be correct
4. Set the `z_adjust` value to the baby stepped Z offset value
5. Restart Klipper

markniu, 2024-04-30:
the z_offset also works well now then no need to set z_adjust manual.