markniu/Bed_Distance_sensor #82: Bed Distance Sensor data error:10.24

kodipa, 2024-01-30:
receive this error from klipper
installed on mks sgen 
sda p1.25
scl p2.0

someone can help me?

markniu, 2024-01-31:
Hello, all the endstop pins for MKS board can not be used, for example the P1.25 is the X+ that is connected to a capacitor that can filter the digital signal from the  BDsensor.  needs to find another pin for the sda.

Townie-au, 2024-02-07:
Hi Mark
I'm also getting this error. Any ideas?

[BDsensor] 
#sda_pin: PB1 # Don't use aliases for the board pins 
#scl_pin: PB0 # example of connecting to main board Creality V4.2.7
scl_pin:MKS_THR:gpio11 # example of connecting to CAN module like MKS THR42 
sda_pin:MKS_THR:gpio21 # example of connecting to CAN module like MKS THR42 
delay: 20 # 20us per pulse, this value should be >=18 but must be below 30
z_offset:0 # this `z_offset` must be set to 0. 
z_adjust:0.0 # z axis adjustment, replace the z_offset function. within -0.3 to 0.3mm
x_offset: 27.8
y_offset: 6.75
no_stop_probe: # enable this for fast probe, the toolhead will not stop at the probe point.
position_endstop: 0.8 # this value must >0.5, the Z axis will be triggered after it measured the distance is below this value(mm) while homing z, recommend value is 0.5~1.0
#speed:0.8 # this speed only works for the z tilt and PROBE_ACCURACY command.

markniu, 2024-02-08:
the data error 1024 means there is a connection problem, you can check the connection,pin name,choose the pins in the EXP1 connector for the BDsensor.

> Hi Mark I'm also getting this error. Any ideas?
> 
> [BDsensor] #sda_pin: PB1 # Don't use aliases for the board pins #scl_pin: PB0 # example of connecting to main board Creality V4.2.7 scl_pin:MKS_THR:gpio11 # example of connecting to CAN module like MKS THR42 sda_pin:MKS_THR:gpio21 # example

Townie-au, 2024-02-09:
I don't have a exp 1 port unfortunately. Can a USB to gpio adapter be used to connect it too?

Townie-au, 2024-02-09:
I'm using a qidi tech X plus 3 which uses a modified MKS THR42
The board is a modified MKS SKIPR. 
The MKS THR has a 3d touch and Z+ port which I'm trying to use.
The only available port on the main board is a Z- port.
The 3D touch port doesn't have a cap to ground but Z+ does. 
The Z- I'm not sure because I only have the pin diagram not the schematic.

Townie-au, 2024-02-09:
[X-6_V1.0_001_PIN.pdf](https://github.com/markniu/Bed_Distance_sensor/files/14218343/X-6_V1.0_001_PIN.pdf)
Can I use H UART 1?

Townie-au, 2024-02-10:
I don't get this error anymore. Was because I didn't rename the bin file correctly when flashing the MCU. 

Still can't get it to work though. I have both BD sensor and klipper labelled dirty.
I tried using PC1 and PC3 but same issue. M102 S-1 blank. 
M102 s-6 incorrect distance from bed even though I moved it closer. Reading always 1024. 

Going of your posts and my results. The only way I can see this working is to change the main board and tool head board completely. 

Really disappointed. I had this on my Ender 3 with a SKR MINI E3 V3 and a BTT EBB42. Worked amazing. 

Let me know if you can think of something to get it working.

markniu, 2024-02-10:
there are capacitors in the z+- on the skipper board,so I think this modified board is also the same.
are there any other pins on the modified MKS THR board?
> I'm using a qidi tech X plus 3 which uses a modified MKS THR42 The board is a modified MKS SKIPR. The MKS THR has a 3d touch and Z+ port which I'm trying to use. The only available port on the main [...]

Townie-au, 2024-02-10:
> there are capacitors in the z+- on the skipper board,so I think this modified board is also the same. are there any other pins on the modified MKS THR board?

No there isn't unfortunately. 
See attached. Green port is part cooling.
[image]

Townie-au, 2024-02-12:
If I add a raspberry pi Pico and used the gpio on that would it work?

Billy-d-Kidd, 2024-03-11:
Just updated the MCU with the instructions here https://github.com/markniu/Bed_Distance_sensor/wiki/Flash-BDsensor(V1.1)-firmware, and now I'm getting the same error message.

I also can not get any info from the sensor now when I do the M102 S-1.

markniu, 2024-03-12:
> If I add a raspberry pi Pico and used the gpio on that would it work?
> 
> Or last resort I could desolder the cap.

we have tested that can work with the RP2040, for example MKS THR42, or you can just use the gpio on the host raspberryPi. https://github.com/markniu/Bed_Distance_sensor/issues/101#issuecomment-1973140398

markniu, 2024-03-12:
> Just updated the MCU with the instructions here https://github.com/markniu/Bed_Distance_sensor/wiki/Flash-BDsensor(V1.1)-firmware, and now I'm getting the same error message.
> 
> I also can not get any info from the sensor now when I do the M102 S-1.

you can check again if it updated the firmware success and no other error message while flashing. https://github.com/markniu/Bed_Distance_sensor/issues/74#issuecomment-1931326433

Billy-d-Kidd, 2024-03-13:
I'm not sure but I now think it may be a problem with flashing, so I have re-directed this problem to:
https://github.com/markniu/Bed_Distance_sensor/issues/96