markniu/Bed_Distance_sensor #83: No resolution on calibration

nullifyr, 2024-02-03:
I have the first version of the sensor about .2mm away from the bed, it is PEI.
I can't get the sensor to exit triggered state after calibration. It is clean.
```
1.6mm
M102 S-2
G91
G1 Z+10 F2940
G90
1.6mm
M102 S-2
G91
G1 Z+1 F2940
G90
G91
G1 Z+1 F2940
G90
1.6mm
M102 S-2
234
233
233
233
234
234
233
233
233
233
233
233
233
234
234
233
233
233
233
233
233
234
233
233
233
233
233
233
233
233
233
233
233
233
233
233
233
233
233
233
M102 S-5
You can send command BDSENSOR_READ_CALIBRATION to check the calibration data
Calibrate Finished!
Please Wait...
Calibrating, don't power off the printer
M102 S-6
```

Is there anything I can do to get this to work or is it defective?

markniu, 2024-02-03:
what's the string returned after M102 S-1?
Could you make sure if the Z axis moves up slowly while calibration?

nullifyr, 2024-02-03:
M102 S-1 returns
`V1.0 pandapi3d.com`
The Z axis is moving upwards.
If I put the nozzle against the bed after calibration I get `0.0mm`
If I move the bed so the sensor is not over it anymore, I get `1.6mm'

markniu, 2024-02-04:
Could you send me a message via the email ? this is the first version, I would like to send you a new version. my email is niujl123@sina.com

> M102 S-1 returns `V1.0 pandapi3d.com` The Z axis is moving upwards. If I put the nozzle against the bed after calibration I get `0.0mm` If I move the bed so the sensor is not over it anymore, I get `1.6mm'

nullifyr, 2024-02-23:
I got the replacements today, two different versions. Thank you!
I hooked up the version that plugs directly into the EXP1. This appears to be the newest version. I fastened the sensor to the toolhead so it was level with the bed.

And got exactly the same result, calibration numbers do not rise. They don't start below 100, though the sensor is 0.5mm away from the bed, set with a razor blade. This version, the numbers start at 230, like the original. 
If I put the sensor next to the bed, I get 0.0mm on S-2. 
When I move the bed so the sensor is over open air, I get 0.4mm.

So I tried the earlier version with the BLtouch style connector. Got that in place, glued. Wired up. 
And got exactly the same result. This version produces calibration numbers that stay at 425 the whole way up.

So, that seems pretty definitive. I am doing something wrong. Or, the sensor does not like a highly magnetic bed. Or, I somehow managed to get 3 faulty sensors.

Is there any way to manually check the sensor with the newest version?

markniu, 2024-02-23:
what type of the highly magnetic bed?  you can try to calibrate it far way the magnetic bed and just put the metal plate bellow the sensor

nullifyr, 2024-02-23:
It is a FulaBed, 6x6 5x10x3mm magnets. 
I tried your idea, I put the flex plate on a box about 100mm tall.

```
V1.2c pandapi3d
M102 S-2
0.2mm
M102 S-6
Calibrating, don't power off the printer
Please Wait...
Calibrate Finished!
You can send command BDSENSOR_READ_CALIBRATION to check the calibration data
M102 S-5
448
451
454
450
456
454
453
457
461
454
453
453
448
455
453
458
450
454
455
455
454
454
456
463
451
456
456
458
454
455
458
455
457
456
456
459
455
455
455
459
M84
M102 S-6
Calibrating, don't power off the printer
Please Wait...
Calibrate Finished!
You can send command BDSENSOR_READ_CALIBRATION to check the calibration data
M102 S-5
465
458
458
459
451
458
456
455
454
456
459
458
453
457
457
458
458
457
457
461
453
456
459
461
452
457
458
451
456
457
458
460
457
456
457
457
459
458
458
462
###This is 4mm off the flex plate
M102 S-2
2.2mm
###ok I manually move the Z back to nearly touching the flex plate
M84
M102 S-2
0.1mm
### now I move the Z up about 2mm
M102 S-2
1.1mm
### so it is measuring something?????? but not accurately
## ok we will try against a block of 5mm thick aluminum
M102 S-6
Calibrating, don't power off the printer
Please Wait...
Calibrate Finished!
You can send command BDSENSOR_READ_CALIBRATION to check the calibration data
M102 S-5
462
453
454
451
451
454
454
458
457
456
455
455
454
455
456
458
454
457
457
453
454
456
455
460
462
456
454
455
462
456
456
454
453
455
456
457
462
458
456
454
M102 S-2
### again 4mm up like it should be
0.7mm
M84
### manually move z back to nearly touching block
M102 S-2
0.0mm
M102 S-2
### manually move z up pretty far, 30mm or more, I didn't measure...
0.8mm
### now I am just even more confused
```

nullifyr, 2024-02-23:
Oh...
Wow. I am very stupid. 
[image]
Look closely, you will see what I have been doing wrong for a whole month.

nullifyr, 2024-02-23:
`M102 S-5
368
369
399
434
467
497
528
556
588
614
642
669
692
716
740
760
782
800
819
839
853
870
887
898
909
922
936
946
955
958
1015
1015
1015
1015
1015
1015
1015
1015
1015
1015`

Success... I am so sorry for wasting your time Mark. 

To anyone else with an issue where you can't seem to get the calibration numbers to change... you might want to make sure the sensor head is oriented downwards. The top of the sensor is where the epoxy is.

nullifyr, 2024-02-23:
Solved. User was stupid, all versions of the sensor work fine. Magnets or no magnets.

closed 2024-02-23 by nullifyr

markniu, 2024-02-23:
oh, as far as I know about 2 people misunderstand this.  We will add a label on the sensor surface next

> `M102 S-5 368 369 399 434 467 497 528 556 588 614 642 669 692 716 740 760 782 800 819 839 853 870 887 898 909 922 936 946 955 958 1015 1015 1015 1015 1015 1015 1015 1015 1015 1015`
>
> Success... I am [...]