markniu/Bed_Distance_sensor #84: Pi as secondary MCU: Main process exited, code=killed, status=11/SEGV

CasperVerswijvelt, 2024-02-03:
Hi, I'm trying to get my new bed distance sensor installed, with the sensor connected to my klipper host (Raspberry Pi 3B) as secondary MCU directly.

I successfully 'flashed' the patched firmware to my Raspberry Pi using these instructions: https://www.klipper3d.org/RPi_microcontroller.html.

The host correctly connects to both my CR6 printer (ERA 1.1.0.3 board) and my secondary pi MCU. However, after executing the `M102 S-1` command, I do not get any reply and the `klipper-mcu` service crashes with the following message:

```
Feb 03 16:20:57 mainsailos systemd[1]: Started Starts the MCU Linux firmware for klipper on startup.
Feb 03 16:26:36 mainsailos systemd[1]: klipper-mcu.service: Main process exited, code=killed, status=11/SEGV
Feb 03 16:26:36 mainsailos systemd[1]: klipper-mcu.service: Failed with result 'signal'.
```

This is my klippy log file: 
[klippy (4).log](https://github.com/markniu/Bed_Distance_sensor/files/14152176/klippy.4.log)

Wiring is connected correctly as far as I can see (when manually moving the printhead closer to the bed, I see the light turn on), but the issue occurs even when the sensor is fully disconnected.

I've also enabled I2C in `raspi-config`, but not sure if that is necessary.
Klipper version is v0.12.0-102-g9f41f53c.

Do you know what could be causing this issue? If I have any misconfiguration or if there is something else I can do to provide more info, please let me know.

CasperVerswijvelt, 2024-02-06:
Happy to try debugging this myself too if you can point me in the right direction

markniu, 2024-02-07:
sorry, I did not see your message, I will test with pi later.

markniu, 2024-02-08:
Hello, I just test it with the pi, seems no problem, did the 5V power connect correctly  to the BDsensor, and what's the version on the BDsensor?

CasperVerswijvelt, 2024-02-08:
BDSensor hardware version: 1.3, 2023.12.26
The pi model is Pi 3 model B.

The 5V power is wired correctly. The light on the BDSensor also lights up when approaching the bed:

https://github.com/markniu/Bed_Distance_sensor/assets/25928551/193c9f9e-a17d-4d15-887d-1038d0621a24

The `M102 S-1 `or `M102 S-2` command also gives no reply, it just crashes without any output:

[image]

My CR6 printer is currently flashed with the non-patched version of the firmware, while my RPI secondary MCU is 'flashed' with the modded version, however I had the same result when my CR6 was also flashed with the patched firmware.

Disconnecting the BDSensor entirely from the pi also gives the exact same result: no answer and crash.

I have also tried connecting the BDSensor to my CR6 board itself (I thought I found some free GPIO pins), with the secondary RPI MCU disabled.
When executing the M102 -S1 command, I do get a reply, but it states 10.24 connection error. I assume the [J3 (PA15) and J4 (PB12)](https://github.com/CR6Community/Marlin/blob/7f865970f8fac7b2a6ea67b0b6d78f8efe463e19/Marlin/src/pins/stm32f1/pins_CREALITY_V453.h#L41) are not usable for the BDSensor.

If there is anything I can do on my end to help debug this issue (e.g. some print statements in between the code or something), please let me know. I would love to get this sensor working :)

CasperVerswijvelt, 2024-02-13:
Any update on this?

CasperVerswijvelt, 2024-02-13:
While I don't really know anything about C and have never programmed in it, I tried debugging a bit and came out on the 
`int addr=atoi((char *)args[2]);` line being the one that causes the crash.

Temporarily hardcoding the value to e.g. 1016 makes it so the klipper-mcu process does not crash and I get replies from the g-code commands.
[image].

Please tell me what this line should be or how I can further debug this, as I have no clue on how to further investigate this.

markniu, 2024-02-14:
this code is let the mcu to receive and parse the data from the pi

but I have got this error `b'Got EOF when reading from device'` that same as you after I changing the `int addr=atoi((char *)args[2]); `to `int addr=atoi((char *)args[1]);`

> While I don't really know anything about C and have never programmed in it, I tried debugging a bit and came out on the `int addr=atoi((char *)args[2]);` line being the one that causes the crash.

CasperVerswijvelt, 2024-02-14:
> so seems that there is a protocol problem or something that is not update.

Everything seems to be up to date as far as I can see:
[image]

I also get a warning from `BD_Sensor.c` when running `make flash`, could that be the cause of the issue?
```
pi@mainsailos:~/klipper $ make flash
  Creating symbolic link out/board
  Building out/autoconf.h
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/trsync.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/neopixel.o
  Compiling out/src/pulse_counter.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/spi_software.o
  Compiling out/src/i2c_software.o
  Compiling out/src/thermocouple.o
  Compiling out/src/sensor_adxl345.o
  Compiling out/src/sensor_angle.o
  Compiling out/src/sensor_mpu9250.o
  Compiling out/src/sensor_lis2dw.o
  Compiling out/src/sensor_bulk.o
  Compiling out/src/BD_sensor.o
src/BD_sensor.c: In function ‘command_I2C_BD_send’:
src/BD_sensor.c:418:19: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
  418 |     int addr=atoi((char *)args[2]);
      |                   ^
  Compiling out/src/linux/main.o
  Compiling out/src/linux/timer.o
  Compiling out/src/linux/console.o
  Compiling out/src/linux/watchdog.o
  Compiling out/src/linux/pca9685.o
  Compiling out/src/linux/spidev.o
  Compiling out/src/linux/analog.o
  Compiling out/src/linux/hard_pwm.o
  Compiling out/src/linux/i2c.o
  Compiling out/src/linux/gpio.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/generic/alloc.o
  Compiling out/src/linux/sensor_ds18b20.o
  Building out/compile_time_request.o
Version: v0.12.0-104-g1b24f6a2
  Linking out/klipper.elf
  Flashing
Installing micro-controller code to /usr/local/bin/
Attempting host MCU restart...
```

Is there any way to read debug prints of some kind from the C code as to further debug this issue?

CasperVerswijvelt, 2024-02-14:
I'm also running a 64-bit OS:
```
pi@mainsailos:~ $ uname -m
aarch64
```

markniu, 2024-02-14:
there is no any warning while compile.
```
pi@octopi:~/klipper $ make
  Creating symbolic link out/board
  Building out/autoconf.h
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/trsync.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/neopixel.o
  Compiling out/src/pulse_counter.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/spi_software.o
  Compiling out/src/i2c_software.o
  Compiling out/src/BD_sensor.o
  Compiling out/src/thermocouple.o
  Compiling out/src/sensor_adxl345.o
  Compiling out/src/sensor_angle.o
  Compiling out/src/sensor_mpu9250.o
  Compiling out/src/sensor_lis2dw.o
  Compiling out/src/sensor_bulk.o
  Compiling out/src/linux/main.o
  Compiling out/src/linux/timer.o
  Compiling out/src/linux/console.o
  Compiling out/src/linux/watchdog.o
  Compiling out/src/linux/pca9685.o
  Compiling out/src/linux/spidev.o
  Compiling out/src/linux/analog.o
  Compiling out/src/linux/hard_pwm.o
  Compiling out/src/linux/i2c.o
  Compiling out/src/linux/gpio.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/generic/alloc.o
  Compiling out/src/linux/sensor_ds18b20.o
  Building out/compile_time_request.o
Version: ?-20240214_130722-octopi
  Linking out/klipper.elf
pi@octopi:~/klipper $ 
```

markniu, 2024-02-14:
there is a way to add a debug message in the code, for example :
`output("command_I2C_BD_send mcuoid=%c cur_mm0=%c", args[0],args[1]);` in the function `command_I2C_BD_send`
that output will display the message into the klippy.log

CasperVerswijvelt, 2024-02-14:
Okay, placing this code in the `command_I2C_BD_send` function

`output("ARGS 0=%u 1=%u 2=%u 3=%u 4=%u 5=%u 6=%u 7=%u",args[0],args[1],args[2],args[3],args[4],args[5],args[6],args[7]);`

Gives me the following output:

`mcu 'host': #output: ARGS 0=1 1=4 2=5 3=85 4=3512604320 5=127 6=2269789564 7=85`

I do believe it might have something to do with 32 vs 64 bit, is your Pi by any chance running a 32-bit OS? On 64-bit OS a pointer is 8 bits which is probably why the `(char *)args[2]` cast fails?

CasperVerswijvelt, 2024-02-15:
I have given up on trying to get this to work on a 64-bit OS and installed MainSailOS 32-bit. While it still does not full work as expected, I have got a lot further now.

I will rename this issue so it is clear for anyone else that will try this in the future

markniu, 2024-03-17:
I just have tested this on the RPI with 64 bit os , it worked as expected.

closed 2024-03-17 by markniu