markniu/Bed_Distance_sensor #86: PROBE_ACCURACY always performs one sample only

Life0fBrian, 2024-02-05:
As I could not find any related information or I'm just blind I'll have to ask here for any hints or maybe a solution.

My printer uses the BD sensor for probing and Z homing.
Due to massive upgrades on the Z motion system and some minor issues I wanted to run the `PROBE_ACCURACY` to check the repeatability of this sensor.
However it only performs one sample though it says otherwise.

> probe accuracy results: maximum 0.990000, minimum 0.990000, range 0.000000, average 0.990000, median 0.990000, standard deviation 0.000000
> PROBE_ACCURACY at X:127.000 Y:105.000 Z:1.000 (samples=5 retract=2.000 speed=0.8 lift_speed=0.8)

Same with the defaults where 10 samples should be performed.

As I can't remember having this performed with the old inductive probe I don't know whether this worked in the past or not.

Tested it in sensor nonstop and normal mode with same results.

[klippy.log](https://github.com/markniu/Bed_Distance_sensor/files/14161820/klippy.log)

markniu, 2024-02-05:
Here the BDsensor no need to move the z axis up and down to measure the distance like normal inductive switch sensor, so it will stop at the position and then just read samples.

Life0fBrian, 2024-02-05:
Hi Mark,
thanks for commenting that fast!

But shouldn't it still somehow show ten samples even if they are the same?
Or would I just see a differing maximum and minimum then?

markniu, 2024-02-05:
you can send M102 S-2 to get the sample to see if it accuracy too.

Life0fBrian, 2024-02-09:
Hi @markniu,

I think we found the issue at least in the Klipper_beta branch:

This looks sus: [Bed_Distance_sensor/klipper_Beta/BDsensor.py at da5e8334aac2502f32d837c15a2a0a6d840f6781 · markniu/Bed_Distance_sensor · GitHub](https://github.com/markniu/Bed_Distance_sensor/blob/da5e8334aac2502f32d837c15a2a0a6d840f6781/klipper_Beta/BDsensor.py#L329C30-L329C33)

You are overwriting the z coordinate in the array object and then appending that same object into a collection multiple times. Same as this:

foo = [1, 2, 3]
bar = []
for i in range(0,10):
    foo[2] = i
    bar.append(foo)

print(bar)

Results:

[[1, 2, 9], [1, 2, 9], [1, 2, 9], [1, 2, 9], [1, 2, 9], [1, 2, 9], [1, 2, 9], [1, 2, 9], [1, 2, 9], [1, 2, 9]]

All the z coordinates are 9 because these are all the same array object. Meaning all the other probes in the loop were lost.

You can fix this with:

foo = [1, 2, 3]
bar = []
for i in range(0,10):
    qux = foo.copy()
    qux[2] = i
    bar.append(qux)

print(bar)

Results:

[[1, 2, 0], [1, 2, 1], [1, 2, 2], [1, 2, 3], [1, 2, 4], [1, 2, 5], [1, 2, 6], [1, 2, 7], [1, 2, 8], [1, 2, 9]]

This is quite obvious as the probe always shows the same result that can't be true in reality.
Thank you for fixing that!

Life0fBrian, 2024-02-28:
@markniu did you check this?

Life0fBrian, 2024-03-02:
Thank you!
With the latest code it now behaves as expected:

```
14:49 probe accuracy results: maximum 0.042813, minimum 0.020625, range 0.022187, average 0.037125, median 0.038906, standard deviation 0.006229
14:49 probe at 127.000,105.000 is z=0.039688 (pos:0.799688 - bd:0.760)
14:48 probe at 127.000,105.000 is z=0.041875 (pos:0.791875 - bd:0.750)
14:48 probe at 127.000,105.000 is z=0.034375 (pos:0.794375 - bd:0.760)
14:48 probe at 127.000,105.000 is z=0.036250 (pos:0.796250 - bd:0.760)
14:48 probe at 127.000,105.000 is z=0.038125 (pos:0.798125 - bd:0.760)
14:48 probe at 127.000,105.000 is z=0.040313 (pos:0.800313 - bd:0.760)
14:48 probe at 127.000,105.000 is z=0.042500 (pos:0.802500 - bd:0.760)
14:48 probe at 127.000,105.000 is z=0.020625 (pos:0.800625 - bd:0.780)
14:48 probe at 127.000,105.000 is z=0.042813 (pos:0.792813 - bd:0.750)
14:48 probe at 127.000,105.000 is z=0.034688 (pos:0.784688 - bd:0.750)
14:48 PROBE_ACCURACY at X:127.000 Y:105.000 Z:1.000 (samples=10 retract=2.000 speed=0.8 lift_speed=0.8)
```

closed 2024-03-02 by Life0fBrian