markniu/Bed_Distance_sensor #88: Calibration problem

PLMD86, 2024-02-08:
Hi markniu I'm installing the bd sensor on a vzbot 330awd but I can't complete the calibration. By placing the nozzle on the plate and launching the M102 S-2 command it gives me a distance of 0.0mm, moving the sensor away to the limit that it can read, it always remains at 0.0mm. I mounted the sensor at a distance of 0.5mm from the nozzle. The version is 1.2 pandapid. If I do the calibration it always gives me the same value but the Z axis doesn't seem to move even if from the Klipper panel I see that from 0 it goes up to 3.9

[image]
[image]
[image]
[image]

PLMD86, 2024-02-08:
This is my configuration in the printer.cfg

[safe_z_home]
home_xy_position: 150,150

[bed_mesh]
speed: 120
mesh_min: 20,20
mesh_max: 250,250
probe_count: 6,6
horizontal_move_z: 0.8
algorithm: bicubic

[force_move] 
enable_force_move: true

[BDsensor]
sda_pin: PG14
scl_pin: PG13
delay: 20
z_offset: 0.0
#z_adjust: 0.0
x_offset: 0
y_offset: -23
#no_stop_probe:
position_endstop: 0
speed: 0.8

[stepper_z]
## In M0 position
step_pin: PE2
dir_pin: PC5
#enable_pin: !PF11
microsteps: 32
rotation_distance: 4
endstop_pin: probe:z_virtual_endstop           #make sure to set the correct pin here.
#position_endstop: 0.0
position_max: 410
full_steps_per_rotation: 200
homing_retract_dist: 5.0
homing_positive_dir: false
homing_speed: 5.0
second_homing_speed: 1
[tmc2209 stepper_z]
interpolate: false
uart_pin: PC4
run_current: 1.2
stealthchop_threshold: 0

markniu, 2024-02-11:
Hello, from the data(250 249 250 ...) , it seems that the nozzle is still touch the bed after it goes up from 0.1 to 0.6 while calibration .  so let the nozzle above the bed about 0.1mm and do a calibartion again.