markniu/Bed_Distance_sensor #89: Delta Printer Homing Broken

echterOsti, 2024-02-08:
Im using the current BDsensor_beta, because i saw it's supposed to bring the real time leveling to klipper.

When i home my delta printer after using the 
BDSENSOR_SET REAL_TIME_HEIGHT=0.32
command, my A motor homes DOWNWARD, away from the switch and into the bed.
B and C home correctly.
It seems that the current beta breaks DELTA homing, i suspect it reverses or ignores the motor direction. (if set to ! / reverse in cfg.)

I also can not get the classic mesh from BED_MESH_CALIBRATE to be applied, no matter if i use BED_MESH_PROFILE LOAD=default.

I've attached the printer.cfg and my makro for starting the print:
[petersmacros.cfg](https://github.com/markniu/Bed_Distance_sensor/files/14214884/petersmacros.txt)
[printer.cfg](https://github.com/markniu/Bed_Distance_sensor/files/14214885/printer.txt)

echterOsti, 2024-02-08:
I just checked out the config file and can confirm, motor A is the only one that has its direction reversed in the config file. It seems that the current implementation ignores/ reverses the direction set in printer.cfg after BDSENSOR_SET REAL_TIME_HEIGHT command is issued.

klarborg, 2025-04-23:
Did you find a solution? I have the same problem?