markniu/Bed_Distance_sensor #90: Internal error during connect: invalid syntax (BDsensor.py, line 1212)

LAZER555, 2024-02-13:
Hi, after the last update i have this error. I reinstalled the sensor with the script, but didn't reflash the mcu

> Internal error during connect: invalid syntax (BDsensor.py, line 1212)
> 
> Once the underlying issue is corrected, use the "RESTART"
> command to reload the config and restart the host software.
> Printer is halted

shlo-mix, 2024-02-13:
as a temporary solution, until mark will fix, you can open SSH file explorer and edit BDsensor.py under line 1212
[image]

change to:

[image]

LAZER555, 2024-02-13:
Thanks it worked, fixed it with winscp :P All of this because of one Enter , haha

markniu, 2024-02-13:
thanks! just fix that error by the last commit.

closed 2024-02-13 by markniu