markniu/Bed_Distance_sensor #91: Aluminum bed

limefrog-the-genuine-one, 2024-02-14:
Just one question:

Does this work reliable with a pure aluminum bed?

ptegler, 2024-02-15:
they will work, calibrated to that bed/mass.

I've now had to give up on this variation, as every time I change a print bed surface, i have to re-cal Z as the print surface and the bed mass are not the same for different print bed plates here.  Yes I have mag/flex beds, aluminum main plate etc, but when I pop on a flex sheet with g-10 or PET etc, the thicness of my first layer is toast as this sensor still sees the original mass, even though i may now be a full mm closer to the nozzle..

limefrog-the-genuine-one, 2024-02-15:
Understood, changing surface materials will will need recalibration. Since my bed wont change the surface ( 'pertinax' glued on cast aluminum) i'll give it a try.

Does anybody know, if the detecting distance is reduced to 53% using pure aluminum like it happens on 'normal' inductive probes?

limefrog-the-genuine-one, 2024-02-16:
Okay, so I'll calculate with a reduced detecting distance around 2 mm, deducting my 0.5 mm 'pertinax' sheet gives me 1.5 mm overall distance and about a third of that for mounting tolerance between probe and nozzle tip leaving me with just 1 mm headroom. Sporty - but seems feasible.

limefrog-the-genuine-one, 2024-02-27:
@markniu 

I propose to mention this circumstance and the correction factors for other types of metal in the technical [description](https://github.com/markniu/Bed_Distance_sensor/tree/new?tab=readme-ov-file#the-future-of-3d-printer-bed-leveling)  to avoid ambiguities.

limefrog-the-genuine-one, 2024-02-27:
Just for reference:

`Correction coefficients (approx.)

Iron     1.00
Stainless steel:   0.76
Brass:    0.50
Aluminum:   0.48`