markniu/Bed_Distance_sensor #92: Pi as secondary MCU: segmentatioon fault on homing Z (G28)

CasperVerswijvelt, 2024-02-15:
After banging my head against a wall in #84, I switched to a 32 bit version of MainSailOS, and now I am getting a lot further.

However, when trying to home Z it again gives a segmentation fault in the `klipper-mcu` process.

Console logs:
```
21:19 Klipper state: Shutdown
21:19 G28
21:17 BDsensorVer:V1.2 pandapi3d.com ,switch_mode=0,collision_homing=0
21:17 BDSENSOR_VERSION
21:16 0.03mm
21:16 BDSENSOR_DISTANCE
21:16 0.21mm
21:16 BDSENSOR_DISTANCE
21:16 0.84mm
21:16 BDSENSOR_DISTANCE
21:16 10.15mm BDsensor:Out of measure Range or too close to the bed
21:16 BDSENSOR_DISTANCE
21:16 1015
21:16 1015
21:16 1015
21:16 1015
21:16 1015
21:16 957
21:16 957
21:16 957
21:16 950
21:16 939
21:16 930
21:16 915
21:16 902
21:16 887
21:16 872
21:16 860
21:16 841
21:16 824
21:16 809
21:16 788
21:16 769
21:16 748
21:16 728
21:16 704
21:16 682
21:16 659
21:16 632
21:16 609
21:16 579
21:16 553
21:16 524
21:16 497
21:16 467
21:16 437
21:16 406
21:16 376
21:16 346
21:16 317
21:16 299
21:16 290
21:16 BDSENSOR_READ_CALIBRATION
21:16 You can send command BDSENSOR_READ_CALIBRATION to check the calibration data
21:16 Calibrate Finished!
21:15 Please Wait...
21:15 Calibrating, don't power off the printer
21:15 BDsensorVer:V1.2 pandapi3d.com ,switch_mode=0,collision_homing=0
21:15 BDSENSOR_CALIBRATE
21:15 SET_KINEMATIC_POSITION Z=200
```

<details>
  <summary>Extract from klippy.log</summary>
[... 397 lines cut ...]
#*# x_count = 10
#*# y_count = 10
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = bicubic
#*# tension = 0.2
#*# min_x = 20.0
#*# max_x = 219.97
#*# min_y = 20.0
#*# max_y = 219.97
#*#
#*# [input_shaper]
#*# shaper_type_x = zv
#*# shaper_freq_x = 99.6
#*# shaper_type_y = mzv
#*# shaper_freq_y = 38.6

```

</details> 

I checked out other homing related issues like #65 and #57, but not could find anything helpful there.
I tried loading a bed mesh before homing, but that does not fix the issue.

Any idea on what could be going on here?

CasperVerswijvelt, 2024-02-15:
Added extra logging in most functions as to better see where it might be going wrong

<details><summary>BD_sensor.c</summary>
<p>

```c
// Code for Bed Distance Sensor
// Mark yue<niujl123@sina.com>
// This file may be distributed under the terms of the GNU GPLv3 license.

#include <string.h>
#include <stdlib.h>
#include "board/gpio.h"
#include "board/irq.h"
#include "board/misc.h"
#include "command.h"
#include "sched.h"
#include "autoconf.h" // CONFIG_*
#include "basecmd.h" // oid_alloc
#include "board/gpio.h" // gpio_out_write
#include "board/irq.h" // irq_disable
#include "board/misc.h" // timer_is_before
#include "command.h" // DECL_COMMAND
#include "sched.h" // struct timer
#include "trsync.h" // trsync_add_signal

#define BYTE_CHECK_OK     0x01
#define BYTE_CHECK_ERR    0x00
#define BD_setLow(x)  gpio_out_write(x,0)
#define BD_setHigh(x) gpio_out_write(x,1)

struct endstop {
    struct timer time;
    uint32_t rest_time, sample_time, nextwake,pin_num;
    struct trsync *ts;
    uint8_t flags, sample_count, trigger_count, trigger_reason;
    struct gpio_in pin;
};

enum { ESF_PIN_HIGH=1<<0, ESF_HOMING=1<<1 };

uint32_t delay_m = 20, homing_pose = 0;
int sda_pin = -1, scl_pin = -1, z_ofset = 0;
uint16_t BD_Data;
uint16_t BD_read_flag=1018,BD_read_lock=0;
int switch_mode = 0; //1:in switch mode
struct gpio_out sda_gpio, scl_gpio;
struct gpio_in sda_gpio_in;
static uint_fast8_t endstop_oversample_event(struct timer *t);
static struct endstop e ;

void BD_i2c_write(unsigned int addr);
uint16_t BD_i2c_read(void);

int BD_i2c_init(uint32_t _sda,uint32_t _scl,
    uint32_t delays,uint32_t h_pose,int z_adjust)
{
    output("CASPER DEBUG BD_i2c_init sda=%u scl=%u delays=%u pose=%u adjust=%u 5=%u 6=%u 7=%u",_sda,_scl,delays,h_pose,z_adjust);
    sda_pin=_sda;
    scl_pin =_scl;
    homing_pose = h_pose;
    z_ofset = z_adjust;
    if (z_ofset > 500)
        z_ofset = 0;
    if(delays>0)
        delay_m=delays;
    sda_gpio=gpio_out_setup(sda_pin, 1);
    scl_gpio=gpio_out_setup(scl_pin, 1);

    gpio_out_write(sda_gpio, 1);
[... 371 lines cut ...]
        e.ts = NULL;
        e.flags = 0;
        return;
    }
    e.rest_time = args[4];
    e.time.func = endstop_event;
    e.trigger_count = e.sample_count;
    e.flags = ESF_HOMING | (args[5] ? ESF_PIN_HIGH : 0);
    e.ts = trsync_oid_lookup(args[6]);
    e.trigger_reason = args[7];
    e.pin_num = args[8];
    e.pin =  gpio_in_setup(args[8], 1);
    sched_add_timer(&e.time);
}
DECL_COMMAND(command_BDendstop_home,
             "BDendstop_home oid=%c clock=%u sample_ticks=%u sample_count=%c"
             " rest_ticks=%u pin_value=%c trsync_oid=%c trigger_reason=%c"
             " endstop_pin=%c");

``` 

</p>
</details> 

[klippy.log](https://github.com/markniu/Bed_Distance_sensor/files/14302580/klippy.log)

markniu, 2024-02-16:
I think I find the problem with this pi mcu, please modify this 
```
while (timer_is_before(timer_read_time(), end))
        irq_poll();
```
to
```
while (timer_is_before(timer_read_time(), end));
       // irq_poll();
```

CasperVerswijvelt, 2024-02-16:
That fixed it! Thanks for the support!

closed 2024-02-16 by CasperVerswijvelt