markniu/Bed_Distance_sensor #94: Move out of range error

Liquidus-tech, 2024-02-23:
Hi, I get this error,
 G28 executed, 
QUAD_GANTRY_LEVEL executed,
 BED_MESH_CALIBRATE I receive the following error: `Move out of range`: -2.000 21.800 1.000 [0.000],
 apparently after updating the various entries.
This appeared in the console (it wasn't there before):
BDsensorVer:V1.2 pandapi3d.com ,switch_mode=0,collision_homing=0
I can't understand the problem. 
Can you help me on how  solve it? Thank you in advance

[klippy.log](https://github.com/markniu/Bed_Distance_sensor/files/14390303/klippy.log)

[printer_cfg.txt](https://github.com/markniu/Bed_Distance_sensor/files/14390322/printer_cfg.txt)

markniu, 2024-02-26:
it seems that the x position -2 is not allowed. 
the reason is here: the mesh_min in the section [bed_mesh] is 20 , but the x_offset = 22 in the senction BDsensor.

Liquidus-tech, 2024-02-27:
Thank you very much, now I have another error, I'll try to solve it myself (if I can)

closed 2025-03-11 by markniu