markniu/Bed_Distance_sensor #95: Collision sensing issues with heightmap

BittorB, 2024-02-24:
if i have issues with klipper and the bed leveling because if you activate the heightmap and you home again the nozzle starts to print very bad the first layer (because if the heightmap is 0.2 and homes again with the nozzle increments 0.2) 

other issue is when you are printing with heatbed at 85 and homes again before printing remains totally uncalibrated because the temp drift. 

i think so the collision sensing needs to work only when you calibrate the probe after that maybe disable the colision sensig . its that posible make?

My temporal solution is calibrate the sensor with collision enabled and the disable this feature to avoid this issues.

markniu, 2024-02-26:
I will add a new variable in the printer.cfg for this.

markniu, 2024-03-03:
a new setting `collision_calibrate` is added for choose nozzle collision sensing in BDsensor calibrating.
```
collision_homing:0 #  set it 1 to enable homing with nozzle collision sensing.
collision_calibrate:0 # set it 1 to enable auto calibrate BDsensor with nozzle collision sensing.
```

BittorB, 2024-03-03:
I have tested now the collision collision_calibrate and works well