markniu/Bed_Distance_sensor #96: Cant flash new firmware

Mharasim1, 2024-02-26:
im trying to flash new fimrware to my v1.1 BD senor and im always getting this error 
[image]

markniu, 2024-02-26:
the v1.1 support collision sensing after flashing new firmware.
was there any more message before the "Re-hadshaking...successful"?
> im trying to flash new fimrware to my v1.1 BD senor and im always getting this error ![bd](https://private-user-images.githubusercontent.com/89615269/307686325-5a2ecfb3-9069-4f46-9ac5-62d1a0b3e5cf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg5MzIzMjYsIm5iZiI6MTcwODkzMjAyNiwicGF0aCI6Ii84OTYxNTI2OS8zMDc2ODYzMjUtNWEyZWNmYjMtOTA2OS00ZjQ2LTlhYzUtNjJkMWEwYjNlNWNmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjI2VDA3MjAyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE3NWQ2ZWQyYTliNTgwZWJkY2Q1MmMxOTExODZkYzcwOGI2ZDgyNzc0Y2NiOTQ3ZDcxNjlkNWI3ODYxODcwM2EmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.K4mnrVOLC5j5BOF5nNqJhHNZfl41xbV4UsH3oLcUZqU) i bought it in november last year its a v1.1 does it support collision sensing after update?

Mharasim1, 2024-02-26:
@markniu 
`Checking target MCU ... 
  MCU type: STC15W408AS
  F/W version: 7.2.5T

Current H/W Option:
  . Current system clock source is internal IRC oscillator
  . Current frequency: 30.047MHz
  . Oscillator gain is HIGH
  . Wakeup Timer frequency: 35.375KHz
  . Do not detect the level of P3.2 and P3.3 next download
  . Power-on reset, use the extra power-on delay
  . RESET pin behaves as IO pin
  . Reset while detect a Low-Voltage
  . Thresh voltage level of the built-in LVD : 2.83 V
  . Inhibit EEPROM operation under Low-Voltage  . CPU-Core supply level : 3.70 V
  . Hardware do not enable Watch-Dog-Timer
  . Watch-Dog-Timer pre-scalar : 256
  . Watch-Dog-Timer stop count in idle mode
  . Program can modify the Watch-Dog-Timer scalar
  . Do not erase user EEPROM area at next download
  . Do not control 485 at next download
  . Do not check user password next download
  . TXD is independent IO
  . TXD pin as quasi-bidirectional mode after reset
  . P2.0 output HIGH level after reset
  . Reference voltage: 1251 mV (Range: 1150~1320mV)
  . Testing time: 2023-9-20

  MCU type: STC15W408AS
  F/W version: 7.2.5T

Adjusting frequency ...			[1.000"]
Adjusted frequency: 30.035MHz (0.116%)

Re-handshaking ... Successful			[0.157"]
Current Baudrate: 115200
Erasing MCU flash ...  
Download failed ! (6.90D)(2024-02-26 07:34:59)
`

8bit-boom, 2024-03-02:
Hello, I trying to flash new firmware of STM32 version of bd sensor. In YouTube video there is some device that connect bd sensor and st-link, do I need similar device to flash it or can I flash using cable that's comes with bd sensor?

markniu, 2024-03-03:
> eed similar de

> Hello, I trying to flash new firmware of STM32 version of bd sensor. In YouTube video there is some device that connect bd sensor and st-link, do I need similar device to flash it or can I flash using cable that's comes with bd sensor?

Only the st-link can flash the STM32 version of bd sensor like the in the video, need to solder wires or with the fixture

markniu, 2024-03-14:
you can check again the download hex file. https://github.com/markniu/Bed_Distance_sensor/issues/74#issuecomment-1931326433

> I tried re-flashing the V1.1, and below is the results. MCU type: STC15W408AS F/W version: 7.2.5T
>

Billy-d-Kidd, 2024-03-14:
Re-downloaded the repo as a zip, tried re-flashing the hex file, STC showed same as previously.
Still 'No data from BDsensor,please check connection' on M102 S-1 command.
Sensor LED triggers when testing it.
I am using Creality 4.2.7 with sda_pin: PB1 & scl_pin: PB0.
I have the 5 port probe connector connected via the cable that came with a 3d touch cable to BDsensor.
It worked like this prior to flashing for collison sensing.

markniu, 2024-03-15:
after flashing, there should be some data message from the serial port of bdsensor, also with version text string at the start. 

[image]

Billy-d-Kidd, 2024-03-15:
Here is the results.
[image]

markniu, 2024-04-02:
Hello @Billy-d-Kidd ,  sorry for I forgot this for a long time.  if the bdsensor cannot work after reflash, please send me your order number or shipping address, I would like to send you the latest version. 

> Here is the results.

closed 2024-04-20 by markniu