markniu/Bed_Distance_sensor #97: Bed distance sensor on SB2040

emessage, 2024-02-26:
Has anyone had success using the sensor with an sb2040 v2?  I've tried connecting clk to gpio28 and SDA to gpio289 but when I execute m102 s-1 it says that the sensor is not connected.  I've even tried connecting clk to pwm0 and gpio25 with no success.  But connecting the pins to my skr 3eZ bltouch pins works fine.

markniu, 2024-02-27:
unfortunately, there are capacitors on the gpio28 29 25 , that 100nf capacitor can filter the communication signal. 
[image]

emessage, 2024-02-27:
> unfortunately, there are capacitors on the gpio28 29 25 , that 100nf capacitor can filter the communication signal. [...]

Ok thanks for the info.  That really sucks.

closed 2024-02-27 by emessage