markniu/Bed_Distance_sensor #100: Endstop triggered during retract when it is not triggered.

nullifyr, 2024-02-29:
Mark, I added another z axis motor to my SKR2 and put the gantry on ball screws to get rid of gantry sag that I could not tune away.
Calibration works great. Testing the measurement after calibration, the Z axis is very accurate now.
But now I can't get the Z to home. No matter where I place the gantry, well above the bed or very near it, I get a "Endstop z still triggered after retract" even if the M119 shows the endstop is open when I start homing.
Nothing changed on the printer other than the addition of a 2209 stepper driver
I attached the log with the relevant dump
[sklippy.log](https://github.com/markniu/Bed_Distance_sensor/files/14441199/sklippy.log)
[printerlog.txt](https://github.com/markniu/Bed_Distance_sensor/files/14441200/printerlog.txt)
You can see in the printer log where I calibrate, measure, move the gantry down above the trigger point, measure and check triggers then repeat that after moving the gantry below the trigger point. 
I also have re-cloned the bdsensor repo, made a clean firmware and flashed it.
This is the newest version of your sensor with the long sensor lead that you sent along with my last order. It definitely worked before this change. I believe there is some bug I introduced.

nullifyr, 2024-02-29:
I turned on collision homing and it works now? 
```
G28
BDsensorVer:V1.2c pandapi3d ,switch_mode=1,collision_homing=1
pos:0.80 new:0.87 auto adjust Z axis +0.07mm,Raw data from 748.0 to 755.0
```

markniu, 2024-02-29:
seems it was triggered in the place where the nozzle did not touch on the bed.  
maybe the homing speed and second homing speed of z motor is too slow. 

> I turned on collision homing and it works now?
> 
> ```
> G28
> BDsensorVer:V1.2c pandapi3d ,switch_mode=1,collision_homing=1
> pos:0.80 new:0.87 auto adjust Z axis +0.07mm,Raw data from 748.0 to 755.0
> ```

nullifyr, 2024-02-29:
I raised the homing to 8 and second homing to 0.8, but the nozzle is still ~3mm away from the bed when at Z0

markniu, 2024-02-29:
What's the configuration of the BDsensor , stepper_z  safe_z_home? and they are not in the log file.

> Mark, I added another z axis motor to my SKR2 and put the gantry on ball screws to get rid of gantry sag that I could not tune away. Calibration works great. Testing the measurement after [...]

markniu, 2024-02-29:
try to increase the second homing speed to 3
> I raised the homing to 8 and second homing to 0.8, but the nozzle is still ~3mm away from the bed when at Z0

nullifyr, 2024-02-29:
Second homing at 1.6 seems to have done the trick 
[klippy.log](https://github.com/markniu/Bed_Distance_sensor/files/14441629/klippy.log)
Here is the full Klipper log. 
If I turn off collision homing I am back to the retract error.

nullifyr, 2024-02-29:
Here is the log after I change back to no 
[klippy.log](https://github.com/markniu/Bed_Distance_sensor/files/14441670/klippy.log)

markniu, 2024-02-29:
how about take the bdsensor off the hotend also the whole wires for bdsensor to avoid the noise from other wires , and put the sensor coil close or away from the bed while homing to see if it works as expected.

nullifyr, 2024-02-29:
OK I tried all your suggestions, it did not change the retract message with collision homing off.
The behavior never changes, close to the bed or far away. 
I even put the coil on a 100mm standoff so it was far away from the toolhead. No change there.
I tried the older version of the sensor too. It does the same thing.

nullifyr, 2024-03-02:
Is there any way to get bdsensor to report the state of the endstops before it performs any movement?
I cannot figure out how to fix my problem here. 

Collision homing does work, but it occasionally reads > 0.3mm of adjustment 
is there any way to set a maximum adjustment the collision homing will allow before it fails?
for instance, if there is a glob of plastic on the nozzle and we do collision homing, the adjustment is wild
`pos:0.80 new:1.62 auto adjust Z axis +0.82mm,Raw data from 81.0 to 124.0`
Needless to say this print failed instantly because the nozzle was several layers higher than it should have started at.

markniu, 2024-03-02:
I also tryed your configure and did not met this problem, could you take a video about that homing process?
> OK I tried all your suggestions, it did not change the retract message with collision homing off. The behavior never changes, close to the bed or far away. I even put the coil on a 100mm standoff so [...]

nullifyr, 2024-03-02:
[https://youtu.be/gmcaOjzPdys](https://youtu.be/gmcaOjzPdys)

The gantry never tries to drop down. The Z is immediately triggered. But, you see me move over the trigger range (the move was 10mm) and I did M119, it was open before I pressed home.

Other things I have done 
Corrected the Z motor phase and direction so both motors have the same configuration.
Trammed the bed again just to be sure

markniu, 2024-03-02:
> https://youtu.be/gmcaOjzPdys
>
> The gantry never tries to drop down. The Z is immediately triggered. But, you see me move over the trigger range (the move was 10mm) and I did M119, it was open before [...]

you can replace the BDsensor.py with this new one that I add a debug info into the code,
[BDsensor.zip](https://github.com/markniu/Bed_Distance_sensor/files/14468116/BDsensor.zip)

nullifyr, 2024-03-02:
[klippy (12).log](https://github.com/markniu/Bed_Distance_sensor/files/14468290/klippy.12.log)`

Klipper state: Shutdown

triggered at 10.150 mm !

Endstop z still triggered after retract

home_start:0.800000

home_start:0.800000

BDsensorVer:V1.2c pandapi3d ,switch_mode=1,collision_homing=0

G28`

nullifyr, 2024-03-02:
I tried some more things but still no joy
-Removed TMC Autotuning and KAMP
-Tried different pins for SDA/SCL
-Went back to the old version of the sensor with a short lead for the coil

With the old version I can see the LED while the homing is in process. It never lights up like it does when it is actually triggered.
I am so confused by that.

nullifyr, 2024-03-04:
YES. 
That fixed it.

```
11:04 AM
Z axis triggered at 0.040 mm,auto adjusted by BDsensor.
11:04 AM
home_start:0.100000
11:04 AM
home_start:0.100000
11:04 AM
BDsensorVer:V1.2c pandapi3d ,switch_mode=1,collision_homing=0
```

nullifyr, 2024-03-04:
Could you clarify what the problem actually was?

closed 2024-03-04 by nullifyr

markniu, 2024-03-12:
> YES. That fixed it.
> 
> ```
> 11:04 AM
> Z axis triggered at 0.040 mm,auto adjusted by BDsensor.
> 11:04 AM
> home_start:0.100000
> 11:04 AM
> home_start:0.100000
> 11:04 AM
> BDsensorVer:V1.2c pandapi3d ,switch_mode=1,collision_homing=0
> ```

Hello, just want to know if it is totally fixed the problem in your printer, and that never happened again? did you try other value of this position_endstop ?

nullifyr, 2024-03-12:
Totally fixed the problem. 
Unless I raise the position_endstop to 0.8mm or more. Then the behavior comes back. This is fine for me as the coil is 0.625mm away from the nozzle tip, but I think it would be a problem for anyone who wanted to mount higher than 0.8mm.

markniu, 2024-03-13:
Thanks, that's also confused me for that I have not repeated this problem in myside. I will debug more. but now I think it could be the communication problem by the signal noise on the cable.

here the position_endstop is not the mount height. it just the trigger position that we set to avoid the nozzle crash into bed with higher homing speed.

markniu, 2024-03-22:
> Totally fixed the problem. Unless I raise the position_endstop to 0.8mm or more. Then the behavior comes back. This is fine for me as the coil is 0.625mm away from the nozzle tip, but I think it would be a problem for anyone who wanted to mount higher than 0.8mm.

someone feedbacked me that the latest update of BD_sensor.c had fixed this issue with  position_endstop : 0.8mm, maybe you can try it if you have time and feedback here too. thanks!

nullifyr, 2024-03-23:
```
Klipper state: Shutdown
triggered at 10.150 mm
Endstop z still triggered after retract
home_pos:80
home_pos:80
BDsensorVer:V1.2c pandapi3d ,switch_mode=1,collision_homing=0,collision_cal=0
G28
Klipper state: Disconnect
FIRMWARE_RESTART
Z triggered at 0.700 mm,auto adjusted.
home_pos:70
home_pos:70
BDsensorVer:V1.2c pandapi3d ,switch_mode=1,collision_homing=0,collision_cal=0
G28
```
No change in behavior. I agree it seems like noise, but when the configuration has position_endstop at 0.7mm the sensor works very well. Well enough that I can level the gantry to 0.008mm in 3 tries. Wouldn't noise affect the readings for probes like that?

Could you make a version that raises the Z to some set distance over the trigger height before it tries to home? I know Klipper already has this logic in the homing code, but I can't think of any other way to test this.