markniu/Bed_Distance_sensor #101: Issue with ramps 1.4

luco85, 2024-02-29:
Hi, i tried to install to my printer with ramps 1.4, but klipper crash and shutdown whenr i send any commansd whit M102. I have a bltouch working and no issue when installed bd patch. There is any chance to use with this board or need to use the gpio pin of rpi? In this case can i use any gpio for SDA and CLK?

markniu, 2024-02-29:
you can use the gpio of rpi, #92

luco85, 2024-02-29:
Ok i'll try. Another issue that i had was after add the moonraker update section. After that i had  an UNKNOW state to update section of mainsail. Any info?

luco85, 2024-03-01:
> you can use the gpio of rpi, #92  

It work fine with rpi. For UNKNOW label in update manager do you have any tip?

luco85, 2024-03-04:
I wanted to ask you to do the integration with klipper! good News!
I installed bd sensor to rpa and work fine! very nice device. thank you!

luco85, 2024-03-06:
@markniu hi! Yesterday i tried to use the collision feature. I update bdsensor section with homing and calibration collision set to 1. When i read sensor it respond firmware v1.2, switch mode 0, collision homing 1, collision calibrate 1. When i run calibration it work as before, it not homing but start to calibrate. I updated your code to last and flashed all mcu. I noticed when i run your code to update i have an error. So need to uninstall and reinstall to update

markniu, 2024-03-06:
The respond firmware should be `V1.2c` not `V1.2 ` , else maybe the firmware did not flash successfully.

luco85, 2024-03-06:
I suspected this so yesterday I ordered st link to flash device

luco85, 2024-03-09:
@markniu hi, there is a typo error here
Marco for display distance on Screen
not Marco but Macro ;)

markniu, 2024-03-10:
> @markniu hi, there is a typo error here Marco for display distance on Screen not Marco but Macro ;)

Thank you :)

markniu, 2024-03-28:
the latest BD_sensor.c code on github has fixed the issue on ramps1.4 board.

closed 2024-03-28 by markniu