markniu/Bed_Distance_sensor #104: sequence to calibrate delta printer (Marlin)

Bob-the-Kuhn, 2024-03-07:
Just installed the BD sensor and finally got past the calibration stage.  I'm hoping this helps out those with delta printers.

I'm running a Tevo Little Monster Delta printer.  Homing is via the endstops.  All I use the BD sensor for is to do leveling..

I tried a lot of sequences before I found one that seems to work repeatably:

1. Pull down the head to the print surface
2. Power cycle the printer
3. M420 V3  - so can tell state of printer
4. G29 D - turn off UBL bed leveling system
5. M102 S-1 - just to prove that we can talk to the BD sensor
6. M102 S4 - tell BD sensor to calibrate from 0.0 through 3.9
7. M102 S-6 - start calibration process

---

Here's a pronterface log of the above sequence

```
Send: M420 V3
Recv: 
Recv: Bed Topography Report:
[...]
Recv: Mesh is valid
Recv: Storage slot: 2
Recv: echo:Bed Leveling ON
Recv: ok
[...]
Send: G29 D
Recv: Unified Bed Leveling System v1.01 inactive
Recv: ok
[...]
Send: M102 S-1
Recv: ok
Recv: BD Sensor version:V1.2d pandapi3d
Recv: 
[...]
Send: M114
Recv: X:0.00 Y:0.00 Z:486.35 E:0.00 Count A:453638B:453638C:453638
Recv: ok
[...]
Send: M102 S4
Recv: ok
Send: M102 S-6
Recv: ok
Recv: c_z0:366.92-364.76
[...]
Recv: BD Sensor Calibrating...
[...]
Recv: G1Z0.0, Z:0.00
Recv: w:0, Z:0.00
Recv: G1Z0.1, Z:0.10
Recv: w:1, Z:0.10
Recv: G1Z0.2, Z:0.20
[...]
[...]
Recv: G1Z3.9, Z:3.90
Recv: w:39, Z:3.90
Recv: BD Sensor calibrated.
[...]

```

ptegler, 2024-03-07:
Is this with the oem control board on your TLM?   My board never made it through the initial power up... display was trashed (lines) so everything was built the first time with a duet3D running RepRap.  My DB sensor (year+ old now) was going to goon the TLM, but instead went with the Duet Smart Effector.  switches, inductors, ir, just too many problems. A nd the TLM glass bed doesn't help the BD     :-(

Bob-the-Kuhn, 2024-03-07:
I've managed to destroy & replace almost everything on my TLM.  I use it a lot to test Marlin software changes.   Head crashes and static electricity damage are common in my misdeeds.

I switched to a BTT Octopus when I could no longer get the original controller.

I've switched to a PEI coated magnetic system over the glass bed.  I crashed the head so often that the underlying glass broke which caused the upper surface to not return to the same height after each thermal cycle which made getting a good first layer really miserable.  I got tired of replacing the glass so I bought a stainless-steel disk and put the magnetic PEI system on that.  Much better.

Best thing I ever did to the printer was take the bottom metal plate to a local machine shop and have them enlarge the hole to 355mm.  350mm glass is much easier to get.  Unfortunately, the cheap Chinese 350mm glass was not as thick and rugged as the original Tevo glass.  

I'm also running a Duet Smart Effector.  Much better than the BLTouch.  It's still real pain to dial in .

I'm hoping the BD sensor will make calibration much faster/easier/consistent.  If that works out, I may switch back to the original TLM system.

ptegler, 2024-03-07:
My TLM verts are linear rails.  Magball 640mm arms, Smart Effector, Oribiter extruder on custom flying mount.
SE works perfectly with the RepRap firmware.    Marlin is my Flashforge.   V-Core 500 is Klipper.  BD sensor is collecting dust on the shelf