markniu/Bed_Distance_sensor #105: Can't get collision sensing to work

Obamium69, 2024-03-09:
First of all, I would like to say that the BD-sensor is a fantastic product.
I tried enabling collision sensing but I can't seem to get it working.
Even when I set `collision_homing:1` and `collision_calibrate:1`  it doesn't calibrate is automatically and I have to manually calibrate it (as described here https://github.com/markniu/Bed_Distance_sensor/wiki/Installing-for-Klipper 5. Calibration).

Printer: Voxelab Aquila X2 (Ender 3 v2 clone)
klipper version: v0.12.0-117-g71604b71
moonraker version: v0.8.0-320-g6b1b8c51
M102 S-1 output: `BDsensorVer:V1.2 pandapi3d.com ,switch_mode=0,collision_homing=1,collision_cal=1`

BD-sensor config:
```
[BDsensor] 
scl_pin:PB0
sda_pin:^PB1
delay: 20
z_offset:0
z_adjust:0.0
x_offset: -45
y_offset: -7
position_endstop: 0.8 
no_stop_probe:
collision_homing:1
collision_calibrate:1
```
Do I need to reflash the mcu or am I missing something else?

markniu, 2024-03-10:
yes, need to relfash the mcu. the returned message should be contained `V1.2c` and `switch_mode=1`
` V1.2c pandapi3d.com ,switch_mode=1,collision_homing=1,collision_cal=1`

markniu, 2024-03-10:
right,  https://github.com/markniu/Bed_Distance_sensor/wiki/Flash-BDsensor(stm32)-firmware

> This means I can flash the firmware https://github.com/markniu/Bed_Distance_sensor/blob/new/hex/V1.3_STM32/V1.2stm32_20240211.hex. Is this correct or am I misunderstanding something?

Obamium69, 2024-03-21:
It seems to be working now. Thanks :)

closed 2024-03-21 by Obamium69