markniu/Bed_Distance_sensor #106: Calibration problem

thanatas, 2024-03-10:
I successfuly instaled the bdsensor. I am using klipper.
none of the calibration attemps gives successfuly result. ı was happy with bltouch, ı assume this  sensor will be better.

the result is devastating. while printing ı need to fix z-offset to do get standart result.

Unfotunately, fixing z offset is not stored. how ı can do that?

markniu, 2024-03-11:
could you paste some data returned after M102 S-5 ? especially the first 10 data.

thanatas, 2024-03-11:
Here the result; I also did another calibration which start at 86. Unfortunately it did not also worked. My last logs;

21:54:34 
// probe at 161.181,304.994 is z=0.105000
21:54:34 
// probe at 132.621,304.994 is z=0.100000
21:54:34 
// probe at 104.061,304.994 is z=0.090000
21:54:34 
// probe at 75.501,304.994 is z=0.090000
21:54:34 
// probe at 46.941,304.994 is z=0.080000
21:54:34 
// Mesh Bed Leveling Complete
21:54:34 
// Bed Mesh state has been saved to profile [default]
// for the current session. The SAVE_CONFIG command will
// update the printer config file and restart the printer.
21:55:24 
// Smart Park location: 12.941,12.144.
22:02:50 
$ SET_GCODE_OFFSET Z_ADJUST=+0.05 MOVE=1
22:03:08 
$ SET_GCODE_OFFSET Z_ADJUST=+0.05 MOVE=1
22:23:45 
Done printing file
22:38:06 
$ M221 S102
22:38:09 
$ M102 S-5
153
176
202
229
258
286
315
344
372
402
430
459
487
513
541
568
591
616
639
664
686
704
725
745
762
780
798
813
827
844
858
871
881
894
906
918
928
937
946
946

thanatas, 2024-03-11:
My config is in the following;
I am suspicious about speed!, Is it too much?

[BDsensor] 
sda_pin: P0.10 
scl_pin: P2.0
delay: 20 
z_offset:0 
x_offset: -34
y_offset: -35
no_stop_probe: 
position_endstop: 0.8 
collision_homing:0 
collision_calibrate:0 
speed: 0.8

[bed_mesh]
speed: 200
mesh_min: 10,10
mesh_max: 270,270
probe_count: 10,10
algorithm: bicubic
fade_start: 1
fade_end: 10
fade_target: 0
horizontal_move_z:1

markniu, 2024-03-11:
the reason maybe the calibrate position is not the actual nozzle position because of the X/Y offset.

the z_offset is replaced by the z_adjust, so the z_offset cannot be saved. 
but now consider that your case, and I am testing that recovered the z_offset as normal probe.
I will update that later.

thanatas, 2024-03-12:
Thank you very much, I will try it as soon as possible. I will give the feedback.

thanatas, 2024-03-16:
Hello thank you for all first,  I was back to home, andI did give a try. Unfortunately, there is a problem in the bdsensor.py because klipper rises an error which is probably related to  "endstop_pin: probe:z_virtual_endstop" in Z section. 

The klipper error is : Unknown pin chip name 'probe'

markniu, 2024-03-16:
seems that there is no BDsensor.py in the klipper, you can run again .sh  `~/Bed_Distance_sensor/klipper/install_BDsensor.sh`
> Hello thank you for all first, I was back to home, andI did give a try. Unfortunately, there is a problem in the bdsensor.py because klipper rises an error which is probably related to "endstop_pin: probe:z_virtual_endstop" in Z section.
> 
> The klipper error is : Unknown pin chip name 'probe'

thanatas, 2024-03-16:
Thanks Mark, so far so good now!

I will test it I inform you!

thanatas, 2024-03-16:
Another question, I figure out main problem when I Install the pluggin. The Klipper git repo returns to "Dirty". When I recover it. It removes
related files. 

The klipper error: 

Repo is dirty. Detected the following modifed files: ['scripts/buildcommands.py', 'src/Makefile']

What do you suggest that?

markniu, 2024-03-22:
> Another question, I figure out main problem when I Install the pluggin. The Klipper git repo returns to "Dirty". When I recover it. It removes related files.
>
> The klipper error:
>
> Repo is dirty. [...]

the best way is integrated into official klipper, and I am trying to do that  https://github.com/Klipper3d/klipper/pull/6490

ptegler, 2024-03-22:
regarding 'dirty'...just indicates untracked files in the master.....   'dirty' is not an issue
[https://discord.com/channels/582187371529764864/1133080505005395988/1133080505005395988](https://discord.com/channels/582187371529764864/1133080505005395988/1133080505005395988)

drupi79, 2024-03-29:
so does this mean we can do a probe z offset like we used to do with a BLTouch? this would save me so many headaches I'm having with my z offset right now.

markniu, 2024-03-30:
right
> so does this mean we can do a probe z offset like we used to do with a BLTouch? this would save me so many headaches I'm having with my z offset right now.

8bit-boom, 2024-04-05:
> the z_offset is working now, you can just update the BDsensor.py and restart the klipper, no need to compile and flash the mcu. [8a16426](https://github.com/markniu/Bed_Distance_sensor/commit/8a16426dc0b5f6b52a5d3112ee30a84b028c32d3)

I have a question. I compiled and flash my mcu per instructions and added BD sensor to Update Manager. When there is new update for BD sensor via Update Manager, do I need to compile and flash my mcu again every yime there is update or update through Update Manager is enough?

markniu, 2024-04-05:
Normally just update it via Update Manager and restart the klipper, no need to compile and flash the mcu unless there is a version or protocol error reported by the klipper. 

> I have a question. I compiled and flash my mcu per instructions and added BD sensor to Update Manager. When there is new update for BD sensor via Update Manager, do I need to compile and flash my mcu again every yime there is update or update through Update Manager is enough?

8bit-boom, 2024-04-05:
Great, thank you for quick answer, I really enjoy my BD senosor.

8bit-boom, 2024-04-07:
One more question: is there a way to set different speed for collision calibrate? One that is different from homing speed.

markniu, 2024-04-10:
> One more question: is there a way to set different speed for collision calibrate? One that is different from homing speed.

Seems the only way is to edit the .cfg file for changing the homing speed.

Billy-d-Kidd, 2024-04-11:
While calibrating the sensor, it homes first. Should it retract before homing? When I set the nozzle on the bed, then M102 S-6, it homes X and Y but does not lift the Z axis, leaving huge gouges in the coating on my spring sheet.

markniu, 2024-04-11:
> While calibrating the sensor, it homes first. Should it retract before homing? When I set the nozzle on the bed, then M102 S-6, it homes X and Y but does not lift the Z axis, leaving huge gouges in the coating on my spring sheet.

it homes the axis with G28 command inside, and you can check the `homing_retract_dist` in the` [stepper_z]` that should be >= 5 with the collision sensing feature

Billy-d-Kidd, 2024-04-12:
Added 'homing_retract_dist: 5' in [stepper_z], and it still homed X and Y first. When it moved to the safe_homing position it then lifted up the 5 and set back down. Should it lift the Z axis first, then home X and Y and sit back onto the bed?

8bit-boom, 2024-04-12:
> > One more question: is there a way to set different speed for collision calibrate? One that is different from homing speed.
> 
> Seems the only way is to edit the .cfg file for changing the homing speed.

Thank you for your answer. Shame, I use collision calibrate for automatic calibrate and then home normally, wish they add some option in Klipper, so you don't have to change config file every time. And again love you product, way better than bltouch and piezo sensors that I had.