markniu/Bed_Distance_sensor #108: Weird Motor Sound at Homing with 3 Z-Motors with 1 driver every motor

RicardoHipp, 2024-03-18:
it works, but on z homing the motors make a weird and loud sound.

https://www.youtube.com/watch?v=t8_eU9_tqEI
the first 10 seconds its normally z moving, then z homing, you can hear the different

I use klipper  v0.12.0-125

i used 3 z axis motors.

with my CR touch everything is fine only changing to BDSensor makes a weird sound at homing in Z, but not on normal moving.

i used 3 z axis motors with one driver every motor

I think it's a problem because while the Z axis is moving, the sensor is always measuring, but it's really very loud.

[config-20240318-225751.zip](https://github.com/markniu/Bed_Distance_sensor/files/14642925/config-20240318-225751.zip)

markniu, 2024-03-19:
this happens in some multi z axis printers, the MCU control the multi z axis driver but at same time need to read the bdsensor at very high frequency. 
there are 2 solutions:
1: connect the BDsensor to the different controller, e.g. raspberryPi GPIO or the CAN module.
2: upgrade the firmware  of BDsensor to the latest version that will let the bdsensor works as a switch sensor while homing.

RicardoHipp, 2024-03-20:
Thanks, I flashed the new firmware and now i works.

closed 2024-03-24 by markniu