markniu/Bed_Distance_sensor #109: Question about updating to new firmware version

Carontes, 2024-03-27:
My bed sensor has version 1.1, I would like to update at version 1.3 but I don't understand if I need to have a new version of the bed sensor that came with version 1.2 already preinstalled or if I can update my bed sensor anyway? Do I always have to have the latest version of the firmware before flashing the new one or can I update from version 1 to version 3  jumping version 2, for example? Thank you

markniu, 2024-03-28:
Hello, there are only 3 hardware version: V1.0, V1.1, V1.3. but the firmware version inside are V1.0, V1.1, V1.2
the V1.0 and V1.1 use the STC as its MCU, V1.3 use the STM32 as its MCU, so there are 2 latest firmware V1.1b and V1.2c for the STC and STM32. https://github.com/markniu/Bed_Distance_sensor/tree/new/hex
if you find the nozzle crash the bed sometimes while homing, you can flash it to the latest firmware, else no need to flash to the latest firmware.  
[image]

Carontes, 2024-03-28:
I tried ti follow [this](https://github.com/markniu/Bed_Distance_sensor/wiki/Flash-BDsensor(V1.1)-firmware) guide, I've used the wires included in the box with the bed sensor to connect it to the TTL Serial: Moved the jumper to connect 5V and VCC instead of 3V3, but when I click Download/Program nothing happens and the only sentence I see is `Checking target MCU ... `. I knoe the TTL works because I've used it to flash Openwrt on some routers, so probably I'm doing something wrong

markniu, 2024-03-28:
after click the Download button, and then reboot the bdsensor with power down and power up.

Carontes, 2024-03-28:
The bed sensor is not connected to the printer but directly to the TTL, so I don't know how to make power off and on without unplug the wires from the TTL

markniu, 2024-03-29:
there is a mistake  :),  we can't flash the sensor via the white connector on the sensor. 
we can just unplug/plug the 5v cable to reboot the BDsensor.

Carontes, 2024-03-29:
So I do I have to set the wires? I knew it was my mistake. Sorry but I'm not familiar with it

markniu, 2024-03-29:
Need to solder wires to bdsensor, and plug the wires to the TTL Serial module board.
reboot the bd sensor by unplug and plug the 5V wire into TTL serial module board after clicking the download button. 
[image]

Carontes, 2024-03-29:
I might be stupid and probably I am, but I see no points where I can solder wires on the bdsensor

[image]
[image]

markniu, 2024-03-29:
this is the first batch of v1.1 and there are no solder points outside. please send me your order number from pandapi3d.com, I would like to send you the latest V1.3.

Carontes, 2024-03-29:
Order No. 10540
Placed on Nov 17, 2023
Thank you for your help and time

ptegler, 2024-03-29:
same boat.... have never been able to use mine since purchasing back in 2002!  (need a usable unit!)
[image]

markniu, 2024-03-30:
thanks for purchase the early version, we will send you new one those days.
> same boat.... have never been able to use mine since purchasing back in 2002! (need a usable unit!)
>

ptegler, 2024-03-30:
> thanks for purchase the early version, we will send you new one those days.
> 
> > same boat.... have never been able to use mine since purchasing back in 2002! (need a usable unit!)

now THAT would be fanTASTIC!  thank you @markniu

gerkow, 2024-04-01:
BDsensorVer:V1.0 pandapi3d.com ,switch_mode=0,collision_homing=0,collision_cal=0
Order No. 10365	Placed on Apr 9, 2023
No Update possible?

Raabi91, 2024-04-01:
same for me have the first batch desing cannot flash the sensor

[image]

markniu, 2024-04-01:
thanks for the first batch,  the track number will be send with email.

> same for me have the first batch desing cannot flash the sensor
>

markniu, 2024-04-01:
> I have V1.0 so ordered a SensorM cant wait 2 try collision_calibrate n homing.

got it, thank you very much, will send one more coil&wire with it.

the collision sensing is very useful and I will use this into the real time leveling future.