markniu/Bed_Distance_sensor #111: No data or corrupt data from BDsensor( )

hidara2000, 2024-04-05:
Hi,
I have two of these devices and am getting the same error on both

## WIRING
[image]

## EBB sb2209 from guide
[image]    

[image]    

Servos = PB9
Probe = PB8

## Klipper Settings
```
[BDsensor] 
# Don't use aliases for the board pins
sda_pin: EBBCan: PB9 # from https://github.com/markniu/Bed_Distance_sensor/issues/50
scl_pin: EBBCan: PB8
delay: 20 # 20us per pulse, this value should be >=18 but must be below 30
z_offset:0 #  within -0.6 to 0.6mm
x_offset: 0
y_offset: 25.0
no_stop_probe: # fast probe that the toolhead will not stop at the probe point,disable it by commenting out.
position_endstop: 0.5 #the Z axis will be triggered after it measured the distance is below this value(mm) while homing z, recommend value is 0.5~1.0
#speed:0.8 # this speed only works for the z tilt and PROBE_ACCURACY command.
collision_homing:0 #  set it 1 to enable homing with nozzle collision sensing.
collision_calibrate:0 # set it 1 to enable auto calibrate BDsensor with nozzle collision sensing.
```

## Flashing using canbus result
[image]

## BDsensor.py exists before compiling and flashing clean
[image]

## WHAT I'VE TRIED
- swap   PB8 for  PB9 in klipper settings
- check wires for shorts and continuity - all good 

## Results from commands
```
M119
```
x:open y:open z:TRIGGERED # whether triggered or not
*NOTE: I can see the light triggering to detect metal when I bring a spanner close to the sensor*
    

```
M102 S-1
```
No data or corrupt data from BDsensor( ), Please check connection
    

```
BDSENSOR_VERSION
```
No data or corrupt data from BDsensor( ), Please check connection    

```
BDSENSOR_READ_CALIBRATION
```
No data or corrupt data from BDsensor( ), Please check connection
BDSensor mounted too close! please mountthe BDsensor 0.2~0.4mm higher

hidara2000, 2024-04-05:
Also tried direct connection to MKS v1.3 board and same issue. Bl touch with confi changes works perfectly using the same pins

markniu, 2024-04-05:
Hello, 
Seems you need to switch the SDA and CLK pin in the printer.cfg:
```
sda_pin: EBBCan: PB8
scl_pin: EBBCan: PB9
```

hidara2000, 2024-04-05:
Cheers for the reply. That's what I had originally but changed it as per the comment in this post https://github.com/markniu/Bed_Distance_sensor/issues/50

either way I've tried both

markniu, 2024-04-05:
you can change it back to the correct order and try again.
if still the same,  try to connect the bdsensor to the mainboard for finding out where the problem is.

hidara2000, 2024-04-05:
I've already tried that and connected directly to the SKR v1.3

markniu, 2024-04-05:
the endstop pin on the skr v1.3 for the bltouch doesn't work with bdsensor, you can use the pin in EXP1 or EXP2 for bdsensor.

> SKR v1.3

hidara2000, 2024-04-05:
I have the EBB SB2209 canbus and need to get it working with this. If it needs I2C I found (image below) in the schematics but cant figure out which pins these are.

[image]

markniu, 2024-04-05:
BDsensor use the i2c, but not standard i2c.
I have tested the other version of EBB board that can work with bdsensor with the BLtouch port.
do you check the wires with multimeter? 

> Also just a quick one. The pins for the BDsensor mention sda and clk. Does it use I2c? if so, I wonder if the pins for BLTouch on the EBB SB2209 are compatible?

hidara2000, 2024-04-05:
do you check the wires with multimeter?

I checked for continuity and shorts

markniu, 2024-04-05:
ok, because I have met the connection loosen problem to the mainboard before.
are there any other unused pins on the board?you can try that, but don't use the endstop pins.

markniu, 2024-04-05:
also did the led work? the led on the bdsensor should be light on if it detects the metal.

hidara2000, 2024-04-05:
LED works as expected

markniu, 2024-04-05:
you can also try to increase the delay value in the [BDsensor] section from 20 to 40, that will slow down the data transfer speed.

Terrradives, 2024-04-06:
I also have been having the same symptoms and issues the last few days.  I'm plugged into a octopus 1.1.  pb6 &7 pins.  lights work.  triggered always active.  M102 codes return the same.

hidara2000, 2024-04-10:
Hi @Terrradives ,
Did you manage to get it working?

hidara2000, 2024-04-10:
@markniu  

Any suggestions on how to connect directly to a Pi and if there is a Python script I can run to test the device?

markniu, 2024-04-10:
yes, you can test it with the gpio on the Pi,  connect the bdsensor to the gpio on the PI like gpio17 and gpio27.

[BDsensor] 
....
scl_pin = host:gpio17 
sda_pin = host:gpio27

compile and flash firmware:
https://www.klipper3d.org/RPi_microcontroller.html

https://github.com/markniu/Bed_Distance_sensor/issues/84

hidara2000, 2024-04-10:
Donno if this will help but I added the following lines to the BDsensor.py
``` python
attrs = vars(self)
attrs = ('\n'.join("%s: %s" % item for item in attrs.items()))

self.gcode.respond_info(f"{attrs}")
self.gcode.respond_info("No data or corrupt data from BDsensor(%s), "
                                    "Please check connection"%self.bdversion)
```
and the response was
```
printer: <__main__.Printer object at 0x7f7f9350d0>
config: 
name: BDsensor
z_adjust: 0.0
z_offset: 0.0
position_endstop: 0.5
stow_on_each_sample: True
no_stop_probe: 
collision_homing: 0
collision_calibrate: 0
activate_gcode: 
deactivate_gcode: 
collision_calibrating: 0
switch_mode: 1
bdversion: 
next_cmd_time: 0.0
action_end_time: 0.0
sda_pin_num: PB8
mcu: 
_invert: 0
oid: 2
mcu_endstop: 
_invert_endstop: 0
oid_endstop: 2
endstop_pin_num: PB8
endstop_bdsensor_offset: 0
cmd_queue: 
trapq: 
trapq_append: 
trapq_finalize_moves: 
stepper_kinematics: 
bd_sensor: 
gcode: 
I2C_BD_receive_cmd2: None
gcode_move: 
bedmesh: None
bd_value: 10.24
results: []
finish_home_complete: None
wait_trigger_complete: None
multi: OFF
adjust_range: 0
old_count: 1000
homeing: 0
reactor: 
status_dis: None
_rest_ticks: 0
_trdispatch: 
_trsyncs: [, ]
ncont: 0
z_last: 0
I2C_BD_receive_cmd: 
_home_cmd: 
toolhead:
```

markniu, 2024-04-10:
> Donno if this will help but I added the following lines to the BDsensor.py
> 
that reported all the data is 10.24(`bd_value: 10.24`) , means no data returned from the sda pin.

hidara2000, 2024-04-10:
Interesing....
When I connected the device to the EBB canbus or directly to the SKR v1.3 I got errors

Now when connected directly to the pi I get
```
BDSENSOR_VERSION
```
[image]

markniu, 2024-04-10:
yes, that's weird.

markniu, 2024-04-10:
you can try to test it by connecting it to the pins on the EXP1 of SKR

Terrradives, 2024-04-10:
I have not yet.  I’m kind of hoping there are tests I can rule things out with.  I’ve switched pins I was using on the octopus board, so I don’t think it’s board or pins.  I’ve got a bltouch currently working on those pins.  It’s not wire.  So I’m left kind of thinking how do I verify I’ve got a correctly flashed boards or verify the sensor is fully functional.Sent from my iPhone

hidara2000, 2024-04-13:
**GOT IT WORKING :)**

## What didn't work
1. Made 3 sets of cables from the supplied ribbon. 
Used the proper crimping tool. 
Checked for shorts and continuity. 
None of these cables worked on the BDsensor but all worked for a BL-touch I had lying around
2. Reflashed firmware on EBB and SKR

## What DID work
1. Made a set of cables using some 26AWG silicone cables. 1st set worked on both BL touch and BDsensor

hidara2000, 2024-04-13:
PS for anyone else, here's a mount I made for the CW2 & Stealthburner. You don't need to modify any existing parts.

https://www.printables.com/model/831679-lazy-bd-sensor-adapter-for-stealthburner-voron

markniu, 2024-04-13:
> PS for anyone else, here's a mount I made for the CW2 & Stealthburner. You don't need to modify any existing parts.
> 
> https://www.printables.com/model/831679-lazy-bd-sensor-adapter-for-stealthburner-voron

Thanks for share!

Terrradives, 2024-04-13:
Is there any chance you’d be willing to make a1.5 meter set and ship them to the USA?Sent from my iPhone

closed 2024-04-14 by hidara2000

hidara2000, 2024-04-14:
@Terrradives 
Honestly I would diy this. Just grab the following
- https://www.amazon.com/dp/B08RMQP6YP?ref=ppx_yo2ov_dt_b_product_details&th=1
- https://www.amazon.com/Electrical-26-Flexible-Silicone-different/dp/B07G2LRX68/ref=sr_1_5?crid=2W88Y02YA9KS5&dib=eyJ2IjoiMSJ9.M-xSpfN5-hrlO-gZuBQziJBpnQjTjhgb0YL3kub0BMEqKFId5w0amJq40RH8hTHcOatq9SSlpdG5ZyFTG3EMAmi2ivcoseP1E1zlKJOs4TDDGdj_dFs-occd0sPuToOwj0mkPjicHfxGnEIz8hka4vn9cnWXytrfm9RRuix2V2jfwD7FnHfgfduNPKBHXd66yBIRve0oEu0FPqu2Eh0z6znlS58wxYTpcv-sm7ZgSgIz6UblFUuedaWlr7IaEL66cDWguZNIZb9Hi1Ku__aghjlhh3OG6chmx70y4tMLMK0.6WIZLhGe_t4WfYmKVpaph9igLB6x4FXpG5v0jesykKI&dib_tag=se&keywords=26awg%2Bsilicone&qid=1713060262&sprefix=26awg%2Bsili%2Caps%2C360&sr=8-5&th=1

For each of the 5 wires, cut off the male ends of two pre-crimped wires and solder the length you need in between. The set comes with heat shrink as well :)