markniu/Bed_Distance_sensor #113: re:Beta, after running 'install_BDsensor.sh', firmware compilation fails with Error 255

m3hrz4d, 2024-04-13:
I cloned the repo, then successfully ran 'klipper_Beta/install_BDsensor.sh', but when trying to compile firmware, it fails just before the end with Error 255" Any ideas that could point me in the right direction?
```
make
Using default symbol values (no '/home/sonic/klipper/.config')
Configuration saved to '/home/sonic/klipper/.config'
  Creating symbolic link out/board
Loaded configuration '/home/sonic/klipper/.config'
Configuration saved to '/home/sonic/klipper/.config'
  Creating symbolic link out/board
  Creating symbolic link out/board
  Building out/autoconf.h
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/trsync.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/sdiocmds.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/neopixel.o
  Compiling out/src/pulse_counter.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/spi_software.o
  Compiling out/src/i2c_software.o
  Compiling out/src/thermocouple.o
  Compiling out/src/sensor_adxl345.o
  Compiling out/src/sensor_angle.o
  Compiling out/src/sensor_mpu9250.o
  Compiling out/src/sensor_lis2dw.o
  Compiling out/src/sensor_ldc1612.o
  Compiling out/src/sensor_bulk.o
  Compiling out/src/BD_sensor.o
  Compiling out/src/stm32/watchdog.o
  Compiling out/src/stm32/gpio.o
  Compiling out/src/stm32/clockline.o
  Compiling out/src/stm32/dfu_reboot.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/generic/armcm_boot.o
  Compiling out/src/generic/armcm_irq.o
  Compiling out/src/generic/armcm_reset.o
  Compiling out/src/../lib/stm32f4/system_stm32f4xx.o
  Compiling out/src/stm32/stm32f4.o
  Compiling out/src/generic/armcm_timer.o
  Compiling out/src/stm32/gpioperiph.o
  Compiling out/src/stm32/adc.o
  Compiling out/src/stm32/i2c.o
  Compiling out/src/stm32/spi.o
  Compiling out/src/stm32/sdio.o
  Compiling out/src/stm32/serial.o
  Compiling out/src/generic/serial_irq.o
  Compiling out/src/stm32/hard_pwm.o
  Building out/compile_time_request.o
Too many message ids
make: *** [Makefile:84: out/compile_time_request.o] Error 255
```

markniu, 2024-04-13:
hello, you can try this https://github.com/markniu/Bed_Distance_sensor/issues/47#issuecomment-1825045187

m3hrz4d, 2024-04-14:
that fixed the issue! thank you!

closed 2024-04-14 by m3hrz4d