markniu/Bed_Distance_sensor #116: MCU Protocol error

AlOmaira, 2024-04-14:
Keep getting this error after updating to the latest klipper. If i disable the bed distance sensor klipper works fine. It needs an update firmware for the bed distance sensor i think please help.
[image]

markniu, 2024-04-15:
I think you did not run the BDsensor install script after updating the klipper that will remove the BDsensor patch code and then flash the latest firmware into the MCU.
you can just run the BDsensor install script and compile the klipper.bin and flash the MCU again.

AlOmaira, 2024-04-15:
Can you share the steps please and how to do that.

markniu, 2024-04-15:
install :

```
cd  ~/Bed_Distance_sensor/klipper/
./install_BDsensor.sh
```
compile:
```
cd ~/klipper/
make
```
then flash the firmware of klipper.bin into MCU again

AlOmaira, 2024-04-15:
Oh yes i did that after updating klipper again and still issue not fixed. I had this same issue arise when i installed an adxl but fixed the issue by updating the adxl firmware in dfu mode. I guess the probe needs a newer firmware to work with the updated klipper

markniu, 2024-04-15:
this error is from the BD_sensor.c which will be compile into the klipper.bin for the MCU, not the firmware of probe. 
you can check again if there is BD_sensor.c file in the klipper/src folder before compile the klipper.bin, and if it is flashed success again.
[image]

markniu, 2024-04-17:
oh, sorry for replay late, did you fix this error? seems that the BDsnesor is connected to the MCU, so you need to flash the mcu.

dhoard, 2024-04-23:
The install script patches the Klipper source code.

1. run the install
2. build the Kipper firmware
3. flash the printer MCU board

MEElms, 2024-07-06:
I'm having the same issue:
I have followed every instruction. Reflashed multiple times, only built with ./make_with_bdsensor.sh ...
Confirmed bd_sensor.c file is in the ~/klipper/src directory, as well as the bd_sensor.py file is in the /klipper/klippy/extras directory.
I have found no solutions to fix this...

Steps taken:
1. Remove all Klipper files, including moonraker, mainsail, etc.
2. Download bd_sensor files
3. Re-download klipper and all requisite packages (moonraker, mainsail, etc.)
4. Install bd_sensor
5. Move to klipper directory
6. make menuconfig
7. ./make_with_bd_sensor.sh
8. ./flash-sdcard.sh /dev/serial/by-id/... creality-v4.2.7
9. Echoed a successful flash with MCU
10. power cycle printer
11. ./flash-sdcard.sh -c ... ... 
 Receive "Dirty" error...
12. Repeat until insanity sets in (current state)

[image]

markniu, 2024-07-09:
please check the version by run `make `after install_BDsensor.sh instead of `./make_with_bd_sensor.sh`
there will be a date in the version.

```
~/Bed_Distance_sensor/klipper/install_BDsensor.sh
cd ~/klipper/
make
./flash-sdcard.sh ..
```
[image]

MEElms, 2024-07-12:
Hey Mark!

I figured out my problem. my creality v4.2.7 board was not flashing the firmware. Since the board does not have a reset button or a way to force the bootloader through bridging pins, it was not writing the firmware.bin to update the original Klipper install. 

I ended up buying an SKR mini E3 v3.0 and installing it (Was looking for an excuse to replace the stock board anyway). Now, everything is working flawlessly. I appreciate your reply and your continued push to get this sensor supported natively within Klipper, as I feel it is a substantial improvement over a BLTouch. 

I was able to run a 10x10 bed mesh in about a minute. Thank you again.