markniu/Bed_Distance_sensor #117: No data or corrupt data from BDsensor(), Please check connection

Townie-au, 2024-04-17:
My setup was broken. So i had to reinstall everything.
I am having an issue with katapult not being able to access after flashing the ebb42.
So I have to flash klipper via stm32 programmer.
I connected bdsensor to the ebb42 when flashing klipper after installing bdsensor.
Now I get this error all the time.
Im using BD Sensor v1.
Everything was working including the bd sensor yesterday.
[klippy (27).log](https://github.com/markniu/Bed_Distance_sensor/files/15009506/klippy.27.log)
[printer.txt](https://github.com/markniu/Bed_Distance_sensor/files/15009519/printer.txt)

Townie-au, 2024-04-17:
Ive checked continuity on the lead from the sensor to the board. OK
Canboard communicating via CAN0.

Townie-au, 2024-04-18:
Is the bd sensor dead? Please help

markniu, 2024-04-18:
you can test BDsensor with the gpio on the Pi.

connect the bdsensor to the gpio on the PI like gpio17 and gpio27.

[mcu host]
serial: /tmp/klipper_host_mcu

[BDsensor]
....
scl_pin: host:gpio17
sda_pin: host:gpio27

compile and flash firmware:
https://www.klipper3d.org/RPi_microcontroller.html

https://github.com/markniu/Bed_Distance_sensor/issues/111#issuecomment-2046685001

Townie-au, 2024-04-18:
Ok. 
Got it working somewhat by flashing the beta firmware.
M102 s-1 returns nothing
M102 s-6 calibrates successfully
However when I go to home z I get "Bed Distance Sensor data error:10.24"
When I first started klipper up I got "collision_homing in printer.cfg invalid". So I commented out both collison homing lines.

markniu, 2024-04-18:
Glad to hear it works again.

Townie-au, 2024-04-18:
What about this Mark?
However when I go to home z I get "Bed Distance Sensor data error:10.24"

markniu, 2024-04-18:
if M102 s-1 returns nothing or 10.24 means the same connection problem.

markniu, 2024-04-18:
did you modify the cable before or just the software?

Townie-au, 2024-04-18:
> did you modify the cable before or just the software?

Just the software.

markniu, 2024-04-18:
for finding out what's the problem hardware or software, I think you can try the gpio on the pi.

Townie-au, 2024-04-18:
> you can test BDsensor with the gpio on the Pi. for example: [#111 (comment)](https://github.com/markniu/Bed_Distance_sensor/issues/111#issuecomment-2046685001) connect the bdsensor to the gpio on the PI like gpio17 and gpio27.
> 
> [mcu host] serial: /tmp/klipper_host_mcu
> 
> [BDsensor] .... scl_pin: host:gpio17 sda_pin: host:gpio27
> 
> compile and flash firmware: https://www.klipper3d.org/RPi_microcontroller.html

host: Current version v0.12.0-124-g09a78c31
Up-to-date MCU(s):
<none>

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.

mcu 'host': Unknown command: I2C_BD_send

Townie-au, 2024-04-18:
> > you can test BDsensor with the gpio on the Pi. for example: [#111 (comment)](https://github.com/markniu/Bed_Distance_sensor/issues/111#issuecomment-2046685001) connect the bdsensor to the gpio on the PI like gpio17 and gpio27.
> > [mcu host] serial: /tmp/klipper_host_mcu
> > [BDsensor] .... scl_pin: host:gpio17 sda_pin: host:gpio27
> > compile and flash firmware: https://www.klipper3d.org/RPi_microcontroller.html
> 
> host: Current version v0.12.0-124-g09a78c31 Up-to-date MCU(s):
> 
> Once the underlying issue is corrected, use the "RESTART" command to reload the config and restart the host software.
> 
> mcu 'host': Unknown command: I2C_BD_send

I connected the BD Sensor to my main board, recompiled and reflashed both EBB and M5P and got the same result. :(

markniu, 2024-04-18:
how about use the gpio in the EXP1? E.g. PB6 PB7

Townie-au, 2024-04-18:
> how about use the gpio in the EXP1? E.g. PB6 PB7

Im going to try the RPI.

Townie-au, 2024-04-18:
3.2 40 Pins GPIO
When Manta series motherboard work with CB1, 40 pins GPIO is a custom IO arrangement, as shown in the figure below, Pin 7 is CB1's "PC7", and Pin 11 is CB1's "PC14". The GPIO number of CB1 corresponds to (Px–PA)*32 + Pin
For example:
PC7 = (PC–PA)*32 + 7 = 2 * 32 + 7 = 71
PH10 = (PH–PA)*32 + 10 = 7 * 32 + 10 = 234
PC7 is numbered in Klipper as pin: host: gpio71, PH10 as pin: host: gpio234

WTH. Trying to make sense of this. I am using pin 6 and pin 8.

Townie-au, 2024-04-18:
[image]

Townie-au, 2024-04-18:
OMG. Finally worked it out. GPIO78 and GPIO74
Same error on CB1 GPIO.

markniu, 2024-04-18:
> OMG. Finally worked it out. GPIO78 and GPIO74 Same error on CB1 GPIO.

worked on RPI GPIO?

Townie-au, 2024-04-18:
> > OMG. Finally worked it out. GPIO78 and GPIO74 Same error on CB1 GPIO.
> 
> worked on RPI GPIO?

No. Same error. No data or corrupt data from BDsensor(), Please check connection

markniu, 2024-04-18:
what's the hardware version of bdsensor?

Townie-au, 2024-04-18:
> what's the hardware version of bdsensor?

How do I tell?

markniu, 2024-04-18:
there is a version text on the sensor on the other side of the white connector, like v1.3 or v1.1

Townie-au, 2024-04-18:
v1.1

markniu, 2024-04-18:
does the led on the sensor work? it will light on if it detects the metal else off.

Townie-au, 2024-04-18:
> does the led on the sensor work? it will light on if it detects the metal else off.

Yes led lights up when metal is brought nearby

markniu, 2024-04-18:
then from the hardware, we can only to find if the problem is from the dead mcu or the cable. 
do you have one more cable to replace the old one for the bdsensor?

Townie-au, 2024-04-18:
> then from the hardware, we can only to find if the problem is from the dead mcu or the cable. do you have one more cable to replace the old one for the bdsensor?

No I dont. I checked the cable with the multimeter.

markniu, 2024-04-18:
if you have a solder tool you can try to connect to a serial model, there will be many useful data output from the serial com port.

Townie-au, 2024-04-18:
Do I need to open the case for that?

markniu, 2024-04-18:
> Do I need to open the case for that?

No   
[image]

Townie-au, 2024-04-18:
Would a ST LINK work?

markniu, 2024-04-18:
> Would a ST LINK work?

No, the MCU on the v1.1 is not stm32

Townie-au, 2024-04-18:
> > Would a ST LINK work?
> 
> No, the MCU on the v1.1 is not stm32

Would you be able to send me another bd sensor please?
Ill get the USB>UART converter and debug this one for you.

markniu, 2024-04-18:
> > > Would a ST LINK work?
> > 
> > 
> > No, the MCU on the v1.1 is not stm32
> 
> Would you be able to send me another bd sensor please? Ill get the USB>UART converter and debug this one for you.

No problem, send me an email with your shipping address.

Townie-au, 2024-04-18:
Thankyou 👍👍👍
What's your email?

markniu, 2024-04-18:
niujl123@sina.com

Townie-au, 2024-04-20:
I've sent you the email mark and I've ordered the adapter. Should get here Tuesday. 
What commands do I send to retrieve the data?

markniu, 2024-04-20:
> I've sent you the email mark and I've ordered the adapter. Should get here Tuesday. What commands do I send to retrieve the data?

any serial software is ok, like pronterface.  no command needed.

Townie-au, 2024-04-20:
> > I've sent you the email mark and I've ordered the adapter. Should get here Tuesday. What commands do I send to retrieve the data?
> 
> any serial software is ok, like pronterface. no command needed.

Does it need to be plugged into the printer as well or just read it straight from the sensor?

What do I need to look for?

markniu, 2024-04-20:
no need to be plugged into the printer.  there will be some message output after it is powered on, and many raw data output. then we can confirm whether this sensor works well.

Townie-au, 2024-04-25:
64▒$4▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒yy▒4e▒▒▒G▒5▒▒▒55▒䊊▒F▒▒!!1▒▒F▒!▒▒▒▒)▒▒))▒▒▒▒▒N▒▒▒▒▒D▒5ڃ▒%!▒▒▒!▒͂▒▒!▒▒1▒▒1▒▒▒▒5▒1▒!!!▒▒▒▒▒▒▒▒▒▒▒)▒▒▒▒▒▒9▒▒9▒▒▒9▒5▒▒Ń▒▒)▒i▒▒)))9▒▒▒▒▒▒▒▒▒▒)▒▒599▒▒▒▒▒9▒)▒▒▒)▒▒%ձ45)▒▒▒▒▒▒5)▒)▒)5▒▒)▒▒▒▒5▒▒▒5)▒%▒ɱ▒▒ʂ▒▒▒)▒▒▒▒▒▒ł)▒▒▒▒▒▒▒▒▒▒)▒)D▒▒▒)))▒▒▒d▒▒▒d▒▒▒▒▒g▒)▒5▒▒)▒▒▒▒䙂▒d▒▒4▒▒▒))▒)▒)5))i▒▒)▒▒)▒▒▒▒▒▒▒▒₂▒▒▒▒䂂)▒)▒▒▒▒)▒▒▒▒▒▒▒▒ʶ▒▒▒)▒▒)▒▒▒▒▒▒▒▒▒▒▒▒)5▒▒))▒▒5▒▒ʂ▒▒)D▒▒▒▒▒▒)i▒▒▒▒▒▒▒▒5▒▒▒▒▒▒▒▒̂▒䂾5▒▒▒▒▒▒▒▒▒▒▒▒▒▒)▒

markniu, 2024-04-25:
> 64▒$4▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒yy▒4e▒▒▒G▒5▒▒▒55▒䊊▒F▒▒!!1▒▒F▒!▒▒▒▒)▒▒))▒▒▒▒▒N▒▒▒▒▒D▒5ڃ▒%!▒▒▒!▒͂▒▒!▒▒1▒▒1▒▒▒▒5▒1▒!!!▒▒▒▒▒▒▒▒▒▒▒)▒▒▒▒▒▒9▒▒9▒▒▒9▒5▒▒Ń▒▒)▒i▒▒)))9▒▒▒▒▒▒▒▒▒▒)▒▒599▒▒▒▒▒9▒)▒▒▒)▒▒%ձ45)▒▒▒▒▒▒5)▒)▒)5▒▒)▒ [...]

the baud rate should be 115200

markniu, 2024-04-25:
that's the raw data from the mcu, and seems the sensor works well.

Townie-au, 2024-04-25:
> that's the raw data from the mcu, and seems the sensor works well.

So why the error in mainsail mark?

Townie-au, 2024-04-25:
> that's the raw data from the mcu, and seems the sensor works well.

I want it to work again. :(
Where do we go from here?

markniu, 2024-04-27:
I think it only to check the connection/pin number and try to increase the delay value in the BDsensor section.

Townie-au, 2024-05-12:
Hey Mark
Thank you so much for sending the new sensor. 
It worked straight away. With the cable I was using before. I didn't use your one because it's too long just to go to the tool head and I didn't want to cut it. 
Unless it's an intermittent thing. I'll keep you posted.
How do I use the realtime leveling?

closed 2024-05-17 by Townie-au