markniu/Bed_Distance_sensor #121: Metallic mounting consideration?

kitchung, 2024-04-21:
Beacon has a metal "keep out" area in their mounting instructions. Does BD_Sensor have similar considerations?

The mount I am designing has a screw directly centered above the coil and a retaining screw for the 2.55mm hole on the side.

Can I use steel screws or must be non-metal screws?

markniu, 2024-04-22:
there is no that metal "keep out" area beacon because BDsensor use the magnetic core coil. 
[image]

ckiefer, 2025-02-04:
Hi @kitchung 
Interested to know if you could finish this work? Is the mount available for download?

Cheers