markniu/Bed_Distance_sensor #123: Bed_mesh issue after last update

luco85, 2024-04-23:
Hi mark i just update bdsensor and  after i received this error
[image]

luco85, 2024-04-23:
Now work, i had a uppercase  word. Before this error was bed_mesh is not a valid object. Need to delete bede_mesh default section under bottom klipper section save. I save my bed mesh. Maybe the name bed_mesh cause error

qp1306, 2024-04-23:
I have the same error how do we fix it?

markniu, 2024-04-23:
sorry, I just reverted back the last change that will may cause error in some case, if the section [bed_mesh] is below the [BDsensor]

luco85, 2024-04-23:
Search another section that have bed_mesh words an try to delete
Look at botton if your config file where klipper save his section

luco85, 2024-04-23:
@markniu thanks! Considering that somebody (like me) have the bdsensor section in an external file cfg

closed 2024-04-26 by markniu