markniu/Bed_Distance_sensor #124: EBB36 installation

zarpyj, 2024-04-23:
Hello Mark, hope you are fine.
It si not one issue, but I would like to ask you if you can update your documentation by adding all the necessary steps when using the BDSensor connected to one EBB36.
In you doc Install for Klipper, you say :
Compile the firmware.
cd ~/klipper/
make menuconfig
make clean
make
Could you add some screen captue in you doc, will be really usefull for anybody.
You say too Flash the firmware into the MCU or the CANbus toolhead board which the BDsensor connected.
How to do this step, where to go which command to use.

Will help neebe like me but sure also many other people.

Thank you in advance

Jean-Jacques

markniu, 2024-04-24:
thanks for this, I will try to install the EBB36 to a printer and record the steps this weekend.

zarpyj, 2024-04-24:
Thank you in advance Mark, appreciated

markniu, 2024-05-01:
I just updated it in here:
https://www.instructables.com/preview/E64L32XLVMXFDB6/
and about the compile and flash, different board has different configuration in the step make menuconfig and the setp of flash firmware make flash, please read the document of the board from the factory, we didn't change any of that.

benbender, 2024-08-01:
@markniu any chance to get those instructions into this repo (as I'm expecting this to be quite a common setup) and close this issue?