markniu/Bed_Distance_sensor #125: Probe goes nuts

P1t0u, 2024-04-25:
This is my probe: BDsensorVer:V1.1 pandapi3d.com ,switch_mode=0,collision_homing=0,collision_cal=0

It is connected to the BL Touch port on a BTT Manta M5P
[BDsensor] 
sda_pin: PC13 
scl_pin: PC15
delay: 20 
x_offset: -33
y_offset: 0
no_stop_probe: 
position_endstop: 0.5 
collision_homing:0 
collision_calibrate:0 

KLIPPER and the bed sensor code were updated on Apr 16, the MCU was reflashed  at that time.  This has been going on since I installed the probe about a month ago.

It will work for a while then it says it is triggered when it is clearly not.

M119
x:open y:open z:open
G28 Z
triggered at 10.150 mm
warning:triggered at 0mm, Please slow down the homing z speed
Z triggered at 0.000 mm,auto adjusted.
M119
x:open y:open z:open

The probe is actually about 120MM from the bed and the LED is OFF

I will then power cycle the printer and all is good for a while; then it will go nuts again

markniu, 2024-04-26:
did you calibrate the bdsensor? and what's the data returned after M102 S-5 ?

P1t0u, 2024-04-26:
Yes I ran M102 S-6, here is the M102 S-5 data:
1015
1015
1015
1015
1015
976
976
976
964
943
921
902
880
857
837
817
794
770
749
727
704
681
661
640
614
589
565
541
517
492
469
444
419
369
369
345
315
287
262

The probe is about 2mm from the tip of the nozzle, I tried making the gap smaller in order to get closer to 100 for the first reading but then I did not have 10 between the first and second values.

G1 Z1
BDSENSOR_DISTANCE
1.02mm

markniu, 2024-04-27:
you can add `homing_retract_dist:0` in the section stepper_z.
and also attach klipply.log

P1t0u, 2024-04-27:
As requested I added homing_retract_dist:0 in my configuration, a few minutes later I changed the filament in the printer and upon loading the new filament I had the error which I confirmed with:
G28 Z
triggered at 10.030 mm
warning:triggered at 0mm, Please slow down the homing z speed
Z triggered at -0.120 mm,auto adjusted.

The probe is 180mm from the bed and M119 correctly reports x:open y:open z:open

I am attaching klippy.log
[klippy-5.log](https://github.com/markniu/Bed_Distance_sensor/files/15139166/klippy-5.log)

markniu, 2024-04-28:
what's the returned value after M102 S-2 instead of M119, it seems the returned value is -0.12

markniu, 2024-04-28:
I also suspect the electrinic noise from other wires for example the motor or heater.
could you try the 
G28 Z 
G28 Z

P1t0u, 2024-04-28:
G28
triggered at 10.150 mm
warning:triggered at 0mm, Please slow down the homing z speed
Z triggered at 0.000 mm,auto adjusted.
M119
x:open y:open z:open
M102 S-2
BDsensor:Out of measure Range or too close to the bed

Sadly M102 S-2 is useless in this case.

Further testing allowed me to get the error on demand, every time I loaded or unloaded the filament I could not home Z properly.  Right now I am at the point where it seems that the probe cable does not like running in the cable chain alongside the extruder stepper motor cable (I have run many print jobs successfully so I don't think the hot end is involved, the extruder stepper motor gets to spin faster while loading and unloading the filament so that is my number one suspect).  I have installed another cable on the outside of the cable chain and went from 100% failures to 0%.  The sampling is still pretty small so I won't call it solved just yet but it looks much better.

Now I need to figure out a permanent way to affix the probe cable on the outside of the cable chain.

markniu, 2024-04-29:
someone also report this electromagnetic interference recently, so now I just updated the new cable back to the same as first version  bdsensor, that is shielded, twisted pairs and high purity copper wire for preventing EMI. it will be shipped with the bdsensor now instead of the old cables in the stock.
[image]

P1t0u, 2024-04-29:
That sounds really nice…. Can I have one?

markniu, 2024-04-30:
> That sounds really nice…. Can I have one?
yes, please send me your shipping address to my email: niujl123@sina.com

lunatec84, 2024-05-08:
I’ve got the same errors and it’s not the 30mm long cable to the Toolboard of my printer! Meanwhile I’ve ruined 2 Bondtech CHT nozzles, the second 350x350mm not so cheap CR3D PEI flex-plate and wasted about 65€ filament with this sensor! Now this sensor looks like really bad beta crap to me. Three month ago, there are a lot of bugs with CAN toolboard timer to close error messages and false z value readings, I’ve ruined the first PEI flex-plate and switched back to quickdraw probe. Yesterday I gave this sensor another try, flashed the firmware to the v1.3 sensor, and compiled and flashed Klipper on the EBB36 using CAN speed at 1M. Calibrated the probe, scratched the bed with the nozzle while doing a QGL (bug), printed 3 parts with perfect first layers and then, I’m getting the same timer to close error after 6,5 hours and 75% finished print, wasting 640g Fiberlogy PA12CF15. With the collision probing, the first 5 attempts were okay and at the 6. attempt, the nozzle crashed hardly into the printbed and ruined with the force of the 4 Z motors, parts from the quad gantry and toolhead. I’m totally pissed off with this crap buggy sensor. With the quickdraw probe QGL was fine, why the heck is your code when doing a QGL, moving at first z down in the center of the bed and moving in this low height to the QGL corner points and scratching the unleveled bed with the nozzle, instead of moving to a safe z height, moving to the first corner point, measuring, moving z back to save height and doing the 3 remaining gantry level points in the same way? That wasn’t the case with earlier versions. Heaters where off during probing, because it gives me false sensor readings and sensor calibration values where also okay on the four calibration attempts I’ve tried. With the money I’ve wasted meanwhile in destroyed parts from this crap sensor usage,  I’ve could bought 4-5 out of the box working beacon sensors! I don’t think that it is a good idea to include this buggy sensor and code in klipper at this time.

markniu, 2024-05-08:
I am so sorry for that waste and thank you for the support and feedback those helpful information. 
the timer too close should be the new bug caused by the recent code committed that I have not test too much.
and about the QGL there is a new quad_gantry_level_probe:1  that can let the probe up and down while doing QGL.

lunatec84, 2024-05-08:
In January 2024, I also encountered issues where the timer to close error appeared with this sensor. Even with your code patched into Klippy, the error occurred intermittently when the sensor wasn't in use. This isn't a new bug, and I'm not the only one experiencing these errors. Today, I've reduced the power, speed, and acceleration for Z-axis movements to prevent further damage caused by sensor or code failures. However, the QGL problem persists despite enabling QGL collision probing. The code needs to be adjusted to include a configurable save Z height variable for movements between QGL points, as a belt-driven gantry can unevenly move 10-20mm when stepper motors get unpowered, potentially causing damage to the printer bed or other components. Additionally, the QGL code seems to ignore given z_hop values. Another issue I encountered today was a Toolboard MCU move queue overflow in mid-print. I suspect that sensor data is overloading the CAN bus even at 1Mbit on a fast printer. Moreover, the sensor occasionally triggers in mid-air, and its LED flickers when it's in the triggered range. It's possible that my sensor is faulty. I’m thinking about opening the sensor housing and connecting the analog pin from the sensor chip to an oscilloscope for further analysis. It would also be helpful if you could provide another sensor for testing. At some points I was near to the trash bin with this sensor or making a video from the actual status and a report, but the only thing beware me from that, is the point that if it worked correctly some times, it’s doing a really good job and i would see this sensor in the official klipper repo soon, but for now there are definitely too much and too heavy bugs for including it. I would give you further informations, feedback and helping you to get this sensor reliable, because I know that it’s not an easy task, especially when real-time comes into play.

markniu, 2024-05-08:
Thank you so much for far more test , please send me your shipping address with email:niujl123@sina.com, I would like to send you the 2 type of new sensor for test.

> In January 2024, I also encountered issues where the timer to close error appeared with this sensor. Even with your code patched into Klippy, the error occurred intermittently when the sensor wasn't in use. This isn't a new bug, and I'm not the only

closed 2024-06-09 by P1t0u