markniu/Bed_Distance_sensor #128: updated now MCU error

qp1306, 2024-05-04:
Updated today and now have this error
[image]

markniu, 2024-05-04:
That's also needs to compile and flash the firmware into MCU.

Bfritts3, 2024-05-04:
im getting the same thing, but when I compile the new firmware it gives me  "make: *** [Makefile:84: out/compile_time_request.o] Error 255"
[image]
[image]

qp1306, 2024-05-04:
What I did to get things working again I formatted the memory card and loaded mainsail OS then installed GitHub, BDsensor then Kiauh and then flash the MCU. By doing this fixed my issues as above

markniu, 2024-05-05:
about too many IDs see here https://github.com/markniu/Bed_Distance_sensor/issues/47

Billy-d-Kidd, 2024-05-11:
Day 3 of dealing with this, and it's starting to get a little bit frustrating now.
I have tried several methods to get this installed and so far none have worked.
I even tried qp1306's method, I guess that was a once off work around for their system, it didn't work for mine.
I checked to make sure all the files were in the correct locations, BD_sensor.c was in /klipper/src and BDsensor.py was in /klipper/klippy/extras. I checked the Makefile in src, for src-y += BD_sensor.c. It was there and had 2 space characters at the end of it. Tried it with them, and without them and still get the same I2C error.
Any other ideas on installing would be a great help. Thanks!

markniu, 2024-05-11:
is there still too many ids error?  if you have compiled the klipper success,then I was suspect it was not flashed success.

> Day 3 of dealing with this, and it's starting to get a little bit frustrating now. I have tried several methods to get this installed and so far none have worked. I even tried qp1306's method, I [...]

Billy-d-Kidd, 2024-05-11:
> is there still too many ids error? if you have compiled the klipper success,then I was suspect it was not flashed success.

I have not received any id errors.

markniu, 2024-05-11:
you can check again if it is flashed success
> I have not received any id errors.

markniu, 2024-05-13:
### mcu 'mcu': Unknown command: I2C_BD_send or mcu 'mcu': Unknown command: Z_Move_Live
This error is from the firmware of mcu, need to compile and flash the new firmware into the mcu
```
cd ~/klipper/
make menuconfig
~/Bed_Distance_sensor/klipper/install_BDsensor.sh
make flash
```

Make sure there is `BD_sensor.o `on there while `make` , else run again `~/Bed_Distance_sensor/klipper/install_BDsensor.sh`
and note `make flash` is the command to flash the firmware into the mcu, this command is different for different board.

Townie-au, 2024-05-14:
> ### mcu 'mcu': Unknown command: I2C_BD_send or mcu 'mcu': Unknown command: Z_Move_Live
> This error is from the firmware of mcu, need to compile and flash the new firmware into the mcu
>
> ```
> cd [...]

For further info. My example BD Sensor connected to EBB42
cd ~/klipper/
make menuconfig
(Enter details as normal for EBB42)
~/Bed_Distance_sensor/klipper/install_BDsensor.sh
(Install BD Sensor Successful)
Make Clean
Make
Then flash EBB42

Error gone

Billy-d-Kidd, 2024-05-16:
> Make sure there is `BD_sensor.o `on there while `make` , else run again `~/Bed_Distance_sensor/klipper/install_BDsensor.sh` and note `make flash` is the command to flash the firmware into the mcu, this command is different for different board.

The 'make flash' command is what was trowing me off so bad.
I had to use, './scripts/flash-sdcard.sh /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 creality-v4.2.7' for it to flash my stm32f103 board.
Now it's flashed and the 'BDSENSOR_VERSION' command responds with 'BDsensorVer:V1.2c pandapi3d ,switch_mode=1,collision_homing=1,collision_cal=1'.
So back in business!

Reubertt, 2024-05-27:
From what I noticed in this new script (./make_with_bdsensor.sh) it removes all the information that it is a "Dirty" version of the klipper for the MCU Firmware. The problem is that my klipper instance is Dirty because of the BDsensor and its dependencies and when checking the MCU firmware they become incompatible. I find the idea of ​​removing Dirty from the Klipper firmware interesting, but at the moment this is causing more problems than it solves. I don't know the entire code to know what is or isn't relevant. But I simply ignored the script and it solved my problem.

ptegler, 2024-05-27:
'dirty' simply signifies those apps contain code NOT in the main branch. It has ZERO effect on ops of Klipper/Klipper Screen/etc
ALL of the RatOS users se this with no issues after months of 'dirty' being the 'identifier' to let you know what IS effected or is not standard.

Reubertt, 2024-05-27:
I understand why it gives the 'dirty' flag, I just don't understand why it introduces this script to remove the flag in the 'official' installation guide.This is a function that is clearly still in the development phase and needs to be better designed before being implemented, even markniu was confused at the beginning of this issue why this error was happening. As I said, I think it's interesting to avoid "contaminating" the official Klipper branch, but until then it will take a while.