markniu/Bed_Distance_sensor #131: TMC UART - GSTAT Errors

Reubertt, 2024-05-09:
I have already reported some situations about the case on Facebook, version [9e5f5d2](https://github.com/markniu/Bed_Distance_sensor/commit/9e5f5d267036a029e5e069bdac9f224eaf8402b3) to have resolved the problem. However, I am forwarding the log for version [929f865](https://github.com/markniu/Bed_Distance_sensor/commit/929f86530d2360858cc1fb57e84b9ee320571da4) for better error analysis.

I thought about removing irrelevant parts but was worried about removing anything that I might miss.

[klippy.log](https://github.com/markniu/Bed_Distance_sensor/files/15258801/klippy.log)

Reubertt, 2024-05-09:
The error happens again when stealthchop_threshold is active. Forwarding new log with only log information (different from the other).

[klippy.log](https://github.com/markniu/Bed_Distance_sensor/files/15258937/klippy.8.log)

markniu, 2024-05-09:
Hello, How about increase the endstop sample time? 
[image]

Reubertt, 2024-05-09:
Changing the sample_time to .008 seems to solve the problem (I'll test .007) however I found a way to reproduce my error. I use a similar configuration from [PR20100 ](https://github.com/Pr20100/SOVOL-SV06-Klipper-profile/tree/main/Files) where you abstract certain settings to change the "speed" the printer works. By simplifying this abstraction of my print.cfg I was able to run it with sample_time .006. This means that some part of this code abstraction may be having a conflict. So my problem is apparently resolved, but be warned that very complex abstractions can cause some bugs with BDSensor.

closed 2024-05-09 by Reubertt