markniu/Bed_Distance_sensor #133: Unknown command: Z_Move_Live

drupi79, 2024-05-09:
Klipper  version v0.12.0-179-g434770ea with the latest install of the stable BD Seneor apparently don't get along. I get the following error after flashing my board **mcu 'mcu': Unknown command: Z_Move_Live**

markniu, 2024-05-10:
seems that the mcu didn't flash success or check if there is BD_sensor.o on the SSH while run compile command `make `

LAZER555, 2024-05-13:
i have the same thing, and when i try to make the klipper.bin i get this error

  Preprocessing out/src/generic/armcm_link.ld
  Building out/compile_time_request.o
Too many message ids
make: *** [Makefile:84: out/compile_time_request.o] Error 255

LAZER555, 2024-05-13:
I tried to make without bdsensor, and it worked. It means something in your update was faulty

  Building out/compile_time_request.o
Version: v0.12.0-180-g79930ed9
  Linking out/klipper.elf
  Creating hex file out/klipper.bin

LAZER555, 2024-05-13:
> seems that the mcu didn't flash success or check if there is BD_sensor.o on the SSH while run compile command `make `

there is a BD_sensor.o , but there's an error anyway

fly@flygemini:~/klipper$ make clean
make -j4
  Creating symbolic link out/board
  Creating symbolic link out/board
  Building out/autoconf.h
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/trsync.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/sdiocmds.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/neopixel.o
  Compiling out/src/pulse_counter.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/spi_software.o
  Compiling out/src/i2c_software.o
  Compiling out/src/thermocouple.o
  Compiling out/src/sensor_adxl345.o
  Compiling out/src/sensor_angle.o
  Compiling out/src/sensor_mpu9250.o
  Compiling out/src/sensor_lis2dw.o
  Compiling out/src/sensor_ldc1612.o
  Compiling out/src/sensor_bulk.o
  Compiling out/src/BD_sensor.o
  Compiling out/src/stm32/watchdog.o
  Compiling out/src/stm32/gpio.o
  Compiling out/src/stm32/clockline.o
  Compiling out/src/stm32/dfu_reboot.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/generic/armcm_boot.o
  Compiling out/src/generic/armcm_irq.o
  Compiling out/src/generic/armcm_reset.o
  Compiling out/src/../lib/stm32f4/system_stm32f4xx.o
  Compiling out/src/stm32/stm32f4.o
  Compiling out/src/generic/armcm_timer.o
  Compiling out/src/stm32/gpioperiph.o
  Compiling out/src/stm32/adc.o
  Compiling out/src/stm32/i2c.o
  Compiling out/src/stm32/spi.o
  Compiling out/src/stm32/sdio.o
  Compiling out/src/stm32/usbotg.o
  Compiling out/src/stm32/chipid.o
  Compiling out/src/generic/usb_cdc.o
  Compiling out/src/stm32/hard_pwm.o
  Preprocessing out/src/generic/armcm_link.ld
  Building out/compile_time_request.o
Too many message ids
make: *** [Makefile:84: out/compile_time_request.o] Error 255