markniu/Bed_Distance_sensor #134: make command error

ibrahimhakkicandan, 2024-05-14:
when i try to make command with bdsensor installed it gives "Too many message ids
make: *** [Makefile:84: out/compile_time_request.o] Error 255"

Here is the full log:

pi@mainsail:~/klipper $ make
  Creating symbolic link out/board
  Building out/autoconf.h
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/trsync.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/sdiocmds.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/neopixel.o
  Compiling out/src/pulse_counter.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/spi_software.o
  Compiling out/src/i2c_software.o
  Compiling out/src/thermocouple.o
  Compiling out/src/sensor_adxl345.o
  Compiling out/src/sensor_angle.o
  Compiling out/src/sensor_mpu9250.o
  Compiling out/src/sensor_lis2dw.o
  Compiling out/src/sensor_ldc1612.o
  Compiling out/src/sensor_bulk.o
  Compiling out/src/BD_sensor.o
  Compiling out/src/stm32/watchdog.o
  Compiling out/src/stm32/gpio.o
  Compiling out/src/stm32/clockline.o
  Compiling out/src/stm32/dfu_reboot.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/generic/armcm_boot.o
  Compiling out/src/generic/armcm_irq.o
  Compiling out/src/generic/armcm_reset.o
  Compiling out/src/../lib/stm32f4/system_stm32f4xx.o
  Compiling out/src/stm32/stm32f4.o
  Compiling out/src/generic/armcm_timer.o
  Compiling out/src/stm32/gpioperiph.o
  Compiling out/src/stm32/adc.o
  Compiling out/src/stm32/i2c.o
  Compiling out/src/stm32/spi.o
  Compiling out/src/stm32/sdio.o
  Compiling out/src/stm32/usbotg.o
  Compiling out/src/stm32/chipid.o
  Compiling out/src/generic/usb_cdc.o
  Compiling out/src/stm32/hard_pwm.o
  Building out/compile_time_request.o
Too many message ids
make: *** [Makefile:84: out/compile_time_request.o] Error 255

if i uninstal with the script provided, the make command runs succesfully

markniu, 2024-05-15:
Hello, you can fix this like this https://github.com/markniu/Bed_Distance_sensor/issues/47#issuecomment-1825045187
this is a problem in klipper for that there is only a few ids left now. I am trying to reduce the ids numbers that used by the sensor and will update that to github next.

ibrahimhakkicandan, 2024-05-16:
Thanks it resolved the issue.

closed 2024-05-16 by ibrahimhakkicandan