markniu/Bed_Distance_sensor #135: add bdsensor to creality k1c

alhanon, 2024-05-15:
Any way to install bd sensor in k1 work with klipper can't insall script add files manually to /usr/share/klipper/klippy/extras/
But not work.
Error message
MCU Protocol error

This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).

Your Klipper version is: 09faed31-dirty
MCU(s) which should be updated:
mcu: Current version 1.3.0.40-5-g6977eaff-dirty-20230711_100121-ubuntu
nozzle_mcu: Current version 1.3.0.40-5-g6977eaff-dirty-20230711_100254-ubuntu
leveling_mcu: Current version 1.3.0.40-5-g6977eaff-dirty-20230711_095416-ubuntu
rpi: Current version v0.11.0-372-gb9ad7605
Up-to-date MCU(s):

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.

mcu 'nozzle_mcu': Unknown command: I2C_BD_send

markniu, 2024-05-16:
This error is the mcu, need to compile and flash the new firmware into the mcu which the bdsensor connected.
could the firmware be updated into nozzle_mcu?

alhanon, 2024-05-16:
Klipper in k1 not official klipper its moded for creality k1 mainboard can't compile it if there's any way to add fille and make change manually for file it will be good. This how to add cartographer prob
https://github.com/K1-Klipper/cartographer-klipper/tree/master

https://www.youtube.com/watch?v=eFFhx5NUUrY