markniu/Bed_Distance_sensor #139: bed mesh result weird... is this normal?

tjengbudi, 2024-05-17:
after install bd sensor and place mesh bed for 30 point to check
i got strange result. is this normal?
[image]

tjengbudi, 2024-05-17:
suddenly the bd sensor cannot response. i do not why. now i event cannot get Z HOME... check it first and confirm after 
will report you

tjengbudi, 2024-05-17:
@markniu i check the cable and change the SDA to used EXP1 port and it still cannot receive any data.
what i must to do?

markniu, 2024-05-17:
Both the SCL and SDA pins are changed to EXP1? make sure the 2 GND wires are connected to the mainboard

tjengbudi, 2024-05-17:
the scl still used probe/servos pin.
i checked with m119 the z always open
i put metal and the light in the bdsensor's led is on, but m119 return z
still open

markniu, 2024-05-17:
M102 S-1 return nothing?

tjengbudi, 2024-05-17:
it is return "BDsensorVer:V1.1 pandapi3d.com
,switch_mode=0,collision_homing=0,collision_cal=0"

tjengbudi, 2024-05-17:
M102 S-2 always return " BDsensor:Out of measure Range or too close to the bed"

markniu, 2024-05-17:
Calibrate again

tjengbudi, 2024-05-17:
how to calibrate. i cannot homing for Z. it never detect my bed. and crash after i homing. force manual?

tjengbudi, 2024-05-17:
> Calibrate again

BDsensor is too far from the bed:1015

markniu, 2024-05-17:
> how to calibrate. i cannot homing for Z. it never detect my bed. and crash after i homing. force manual?

yes ,use the manual and SET_KINEMATIC_POSITION , M102 S-6

markniu, 2024-05-17:
and then check the data after M102 S-5

tjengbudi, 2024-05-17:
> > how to calibrate. i cannot homing for Z. it never detect my bed. and crash after i homing. force manual?
> 
> yes ,use the manual and SET_KINEMATIC_POSITION , M102 S-6

after i send M102 S-6 i got BDsensor is too far from the bed:1015

after i send M102 S-5 i got 

"
02:46:26 pm
1015
02:46:26 pm
1015
02:46:26 pm
1015
02:46:26 pm
979
02:46:26 pm
979
02:46:26 pm
979
02:46:26 pm
964
02:46:26 pm
943
02:46:26 pm
923
02:46:26 pm
902
02:46:26 pm
882
02:46:26 pm
860
02:46:26 pm
982
02:46:26 pm
977
02:46:26 pm
950
02:46:26 pm
922
02:46:26 pm
895
02:46:26 pm
907
02:46:26 pm
879
02:46:26 pm
686
02:46:26 pm
663
02:46:26 pm
641
02:46:26 pm
617
02:46:26 pm
594
02:46:26 pm
569
02:46:26 pm
545
02:46:26 pm
519
02:46:26 pm
396
02:46:26 pm
265
02:46:26 pm
587
02:46:26 pm
588
02:46:26 pm
558
02:46:26 pm
528
02:46:26 pm
357
02:46:26 pm
306
02:46:26 pm
280
02:46:26 pm
252
02:46:26 pm
222
02:46:26 pm
195
02:46:26 pm
175
02:46:26 pm
M102 S-5
"

markniu, 2024-05-17:
the data looks right.  btw the led on represent that it is in the measure range but not mean it is triggered for the z axis

tjengbudi, 2024-05-17:
and now it suddenly "BDsensor is too far from the bed:1015" again

markniu, 2024-05-17:
you can comment out the `no_stop_probe` in the section BDsensor, and send M102 S-2 to check the actual distance after it stopped by that error

tjengbudi, 2024-05-17:
after some touch here and touch there... it is working again.
check cable with multitester. it is connected ok

after plug again and test it is run
but the result is amaizing

[image]

M102 S-2 fave result "0.42mm"

markniu, 2024-05-17:
is the data returned by M102 S-2 almost the same after very 
G28  
G1 Z1   
M102 S-2

tjengbudi, 2024-05-17:
after idle... it stop working again... frustrated what happened with this sensor

it crash after i sent g28

tjengbudi, 2024-05-17:
after hard reboot 

03:48:53 pm
0.42mm
03:48:40 pm
G28
G1 Z1
M102 S-2
03:48:32 pm
0.42mm
03:48:19 pm
G28
G1 Z1
M102 S-2
03:48:10 pm
0.41mm
03:47:58 pm
G28
G1 Z1
M102 S-2

tjengbudi, 2024-05-17:
trying check and meassure without changing the height at all

the result very strange "
03:53:00 pm
0.44mm
03:53:00 pm
M102 S-2
03:52:56 pm
0.44mm
03:52:56 pm
M102 S-2
03:52:56 pm
1.03mm
03:52:56 pm
M102 S-2
03:52:55 pm
1.03mm
03:52:55 pm
M102 S-2
03:52:55 pm
1.03mm
03:52:55 pm
M102 S-2
03:52:54 pm
1.03mm
03:52:54 pm
M102 S-2
03:52:53 pm
1.03mm
03:52:53 pm
M102 S-2
03:52:52 pm
1.03mm
03:52:52 pm
M102 S-2
03:52:52 pm
1.03mm
03:52:52 pm
M102 S-2
03:52:52 pm
1.03mm
03:52:52 pm
M102 S-2
03:52:51 pm
1.03mm
03:52:51 pm
M102 S-2
03:52:51 pm
1.03mm
03:52:51 pm
M102 S-2
03:52:51 pm
1.03mm
03:52:51 pm
M102 S-2
03:52:51 pm
1.03mm
03:52:51 pm
M102 S-2
03:52:50 pm
1.03mm
03:52:50 pm
M102 S-2
03:52:50 pm
1.03mm
03:52:50 pm
M102 S-2
03:52:50 pm
1.03mm
03:52:50 pm
M102 S-2
03:52:50 pm
1.03mm
03:52:50 pm
M102 S-2
03:52:49 pm
1.03mm
03:52:49 pm
M102 S-2
03:52:49 pm
1.03mm
03:52:49 pm
M102 S-2
03:52:48 pm
1.03mm
03:52:48 pm
M102 S-2
03:52:48 pm
1.03mm
03:52:48 pm
M102 S-2
03:52:47 pm
1.03mm
03:52:47 pm
M102 S-2
03:52:47 pm
1.03mm
03:52:47 pm
M102 S-2
03:52:46 pm
1.03mm
03:52:46 pm
M102 S-2
03:52:45 pm
1.03mm
03:52:45 pm
M102 S-2
03:52:44 pm
1.03mm
03:52:44 pm
M102 S-2
03:52:43 pm
1.03mm
03:52:43 pm
M102 S-2
03:52:42 pm
1.03mm
03:52:42 pm
M102 S-2
03:52:41 pm
0.99mm
03:52:41 pm
M102 S-2
03:52:29 pm
0.99mm
03:52:29 pm
M102 S-2
"

it can changing by him self from 0.99 --> 1.03 -->0.44

tjengbudi, 2024-05-17:
must i reflash it ?

markniu, 2024-05-17:
I suspend that there is a hardware problem maybe the cable EMI from other wires. where and when do you purchase this v1.1 . I would like to send you a new one, you can send me your shipping address to my email niujl123@sina.com

markniu, 2024-05-17:
did the hotend heat?

> it can changing by him self from 0.99 --> 1.03 -->0.44

tjengbudi, 2024-05-17:
No, cold bed...

tjengbudi, 2024-05-18:
> I suspend that there is a hardware problem maybe the cable EMI from other wires. where and when do you purchase this v1.1 . I would like to send you a new one, you can send me your shipping address to my email [niujl123@sina.com](mailto:niujl123@sina.com)

already sent please take a look