markniu/Bed_Distance_sensor #140: I2CBD error

qp1306, 2024-05-19:
[image]

markniu, 2024-05-19:
need to run
```
cd ~/klipper/
~/Bed_Distance_sensor/klipper/install_BDsensor.sh
 ./make_with_bdsensor.sh 
```
to compile the firmware and flash into the mcu.
here the version is from the github of klipper.

qp1306, 2024-05-19:
Thank you that has fixed my issue.

bombsquad85, 2024-08-29:
> need to run
> 
> ```
> ~/Bed_Distance_sensor/klipper/install_BDsensor.sh
> cd ~/klipper/
> ./make_with_bdsensor.sh 
> ```
> 
> to compile the firmware and then flash into the mcu again. here the version is from the github of klipper.

this didnt work for me. i tried it another way (https://github.com/markniu/Bed_Distance_sensor/issues/145) and it worked, however, i got an error when testing
i am at a loss here

sadgit, 2025-01-21:
> MCU Protocol error
> 
> This is frequently caused by running an older version of the
> firmware on the MCU(s). Fix by recompiling and flashing the
> firmware.
> 
> Your Klipper version is: v0.12.0-410-gcf3b0475-dirty
> MCU(s) which should be updated:
> mcu: Current version v0.12.0-410-gcf3b0475
> EBBCan: Current version v0.12.0-410-gcf3b0475
> Up-to-date MCU(s):
> 
> 
> Once the underlying issue is corrected, use the "RESTART"
> command to reload the config and restart the host software.
> 
> mcu 'mcu': Unknown command: I2CBD

- Tried reflashing both MCU and EBB several times

bombsquad85, 2025-01-21:
> Cleaned out ShakeNTune - but still getting same error but versions now clean
> 
> > MCU Protocol error
> > This is frequently caused by running an older version of the
> > firmware on the MCU(s). Fix by recompiling and flashing the
> > firmware.
> > Your Klipper version is: v0.12.0-410-gcf3b0475
> > MCU(s) which should be updated:
> > Up-to-date MCU(s):
> > mcu: Current version v0.12.0-410-gcf3b0475
> > EBBCan: Current version v0.12.0-410-gcf3b0475
> > Once the underlying issue is corrected, use the "RESTART"
> > command to reload the config and restart the host software.
> > mcu 'mcu': Unknown command: I2CBD

so you are getting the protocol error?

try the new directions in the github. after make clean, use make instead of make w/bdsensor

sadgit, 2025-01-21:
reflashed and now fixed.

bombsquad85, 2025-01-21:
> reflashed and now fixed.

Glad it's working