markniu/Bed_Distance_sensor #141: getting MCU 'mcu' shutdown: Timer too close when running realtime Z Adjust.

drupi79, 2024-05-25:
I'm getting a  MCU 'mcu' shutdown: Timer too close when running real-time Z Adjust. if I go back to just using KAMP or the bed mesh I don't have the issues. this happens with my BDSensor 1.1 and 1.2 running on Ender 3's with 4.2.7 Creality MCU's and Pi 4 4GB

markniu, 2024-05-25:
you can reduce the communication speed by reduce the `delay `value in section [BDsensor] from 20 to 8 for BDsensor V1.2, and also try to slow down the printing speed of first layer or the `max_accel `value.

markniu, 2024-05-25:
here I just added a variables for the sample time,need to update and compile&reflash klipper.bin
[image]

drupi79, 2024-05-26:
so did the update for this and added the rt_sample_time line to my config and now I get a "**Unable to obtain 'I2CBDr' response**" during homing when starting a print.

markniu, 2024-05-26:
seems that you have updated the code, but need to compile and reflash the firmware.

drupi79, 2024-05-26:
you mean flash the BDsensor or klipper? because I reflashed the printer this morning and still have the same error after it homes for printer start

markniu, 2024-05-27:
flash the klipper.bin.  you can try again. maybe you forget to recompile or the flash not success.
the mcu can't recognize this new command rt_sample_time if the firmware is not updated.

markniu, 2024-05-27:
could you check if the BD_sensor.c is the latest as the shown below
[image]

Townie-au, 2024-05-27:
Also try this
My klipper says dirty after installing bdsensor
If it does...
Log in SSH
Run ./Bed_Distance_sensor/klipper/uninstall.sh
(If this doesn't work. Run Sudo chmod +X ./Bed_Distance_sensor/klipper/uninstall.sh)
Sudo rm -r ./Bed_Distance_sensor
Go back to Mainsail 
Do soft recovery on klipper
Make sure everything is updated
Reboot
Now start marks install guide again
When you run make and the list scrolls through make sure BD Sensor is in the list.

Billy-d-Kidd, 2024-05-27:
This script does not seem to like Kiauh. So you will need to flash it at command prompt.

Townie-au, 2024-08-14:
I'm getting this issue again.
I had it working for a bit and then updated klipper, BD sensor, Moonraker etc and now I get it again.
I've added rt_sample time and reduced delay to 8.
The before layer change gcode is in the sliced. 
I've removed all bed mesh related settings and data from my config. 
[Uploading klippy.log (18).txt…]()

Townie-au, 2024-08-14:
[klippy.log (18).txt](https://github.com/user-attachments/files/16617633/klippy.log.18.txt)

markniu, 2024-08-16:
How about the CPU loading while printing with SSH command `htop`?
and maybe need to slow down the printing speed of first layer.

> I'm getting this issue again. I had it working for a bit and then updated klipper, BD sensor, Moonraker etc and now I get it again. I've added rt_sample time and reduced delay to 8. The before layer change gcode is in the sliced. I've removed all bed mesh related settings and data from my config.

Townie-au, 2024-08-16:
> How about the CPU loading while printing with SSH command `htop`? and maybe need to slow down the printing speed of first layer.

Thanks Mark 
What do you mean by htop?

Ok. I'll give that a go.

Townie-au, 2024-08-16:
I slowed down 80%
[image]
[image]
[image]

Townie-au, 2024-08-16:
It's like it's ignoring the disable real time levelling command during print but it's definately in the gcode.

If I enable it manually without printing the bed does move up and down.

Townie-au, 2024-08-16:
[image]

Townie-au, 2024-08-16:
[klippy.log (21).txt](https://github.com/user-attachments/files/16635526/klippy.log.21.txt)

Townie-au, 2024-08-16:
I just did a print of a different gcode and worked correctly.
This wasn't sliced and uploaded recently.

markniu, 2024-08-16:
or continue to increase the value rt_sample_time.  if still happed, you can disable the real time level to see if it caused by it.

Townie-au, 2024-08-17:
I tried all different values for that went up to 32. 
Yes it completed with realtime levelling disabled.
🤔

StiviSVK, 2025-02-27:
Hi any solutions to this problem?

markniu, 2025-02-28:
It will create a new software task in the firmware mcu to read sensor data and adjust the z height if the real time level is enabled.
That task will consume some MCU time, if the time is too long while the printing is too fast, then it will cause this timer too close.

The solution is to increase the interval process time( modify the value `rt_sample_time ` ) or slow down the first layer print  speed.

BTW, This real_time_level runs only in the first layer printing.