markniu/Bed_Distance_sensor #145: mcu 'EBBCan': Unknown command: I2CBD

terri1769, 2024-05-29:
I just had to uninstall and reinstall Klipper to fix a "dirty" issue.  I installed Klipper with KIAUH then went through the install steps for BDSensor.

~/Bed_Distance_sensor/klipper/install_BDsensor.sh
cd ~/klipper/
make menuconfig
./make_with_bdsensor.sh
make flash

When I try to start Klipper I get:
[image]

I thought it might be issue 140 but the solution doesn't work and I don't currently have a "dirty" config.

terri1769, 2024-05-30:
[klippy.log](https://github.com/markniu/Bed_Distance_sensor/files/15501979/klippy.log)

markniu, 2024-05-30:
was there a message `out/src/BD_sensor.o` on the SSH while run make_with_bdsensor.sh?

This error is from the firmware of mcu, and the code of BD_sensor.c does not build into the firmware.

terri1769, 2024-05-30:
Yes.  I built it using those instructions for the Octopus Max EZ board and
then for the EBB36 board then flashed it to each after it was built for
that device.  Here's the text of doing so for the EBB36.

***@***.***:~/klipper $ ./make_with_bdsensor.sh
compiling BD_sensor.c into the klipper firmware
  Creating symbolic link out/board
  Building out/autoconf.h
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/trsync.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/neopixel.o
  Compiling out/src/pulse_counter.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/spi_software.o
  Compiling out/src/i2c_software.o
  Compiling out/src/thermocouple.o
  Compiling out/src/sensor_adxl345.o
  Compiling out/src/sensor_angle.o
  Compiling out/src/sensor_mpu9250.o
  Compiling out/src/sensor_lis2dw.o
  Compiling out/src/sensor_ldc1612.o
  Compiling out/src/sensor_bulk.o
  Compiling out/src/BD_sensor.o
  Compiling out/src/stm32/watchdog.o
  Compiling out/src/stm32/gpio.o
  Compiling out/src/stm32/clockline.o
  Compiling out/src/stm32/dfu_reboot.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/generic/armcm_boot.o
  Compiling out/src/generic/armcm_irq.o
  Compiling out/src/generic/armcm_reset.o
  Compiling out/src/generic/timer_irq.o
  Compiling out/src/stm32/stm32f0_timer.o
  Compiling out/src/stm32/stm32g0.o
  Compiling out/src/stm32/gpioperiph.o
  Compiling out/src/stm32/stm32f0_adc.o
  Compiling out/src/stm32/stm32f0_i2c.o
  Compiling out/src/stm32/spi.o
  Compiling out/src/generic/canserial.o
  Compiling out/src/../lib/fast-hash/fasthash.o
  Compiling out/src/stm32/fdcan.o
  Compiling out/src/generic/canbus.o
  Compiling out/src/stm32/chipid.o
  Compiling out/src/stm32/hard_pwm.o
  Building out/compile_time_request.o
Version: v0.12.0-207-g6cd17420
  Preprocessing out/src/generic/armcm_link.ld
  Linking out/klipper.elf
  Creating hex file out/klipper.bin
Updated 1 path from the index
Updated 1 path from the index

markniu, 2024-05-31:
you still can run the make instead of ./make_with_bdsensor.sh after install_BDsensor.sh, then you can see if the firmware is flashed success into the EBBcan by seeing if there is text dirty on the version page
```
~/Bed_Distance_sensor/klipper/install_BDsensor.sh
cd ~/klipper/
make menuconfig
make
make flash
```

aldoale, 2024-05-31:
I have the same error

Magoo007, 2024-05-31:
[image]
Same errore, the solution here and also in topic #140 doesn’t fix it

markniu, 2024-05-31:
please use the old methond to compile:
```
~/Bed_Distance_sensor/klipper/install_BDsensor.sh
make
```
then if the compile and flash goes well there should be a dirty and compile date in the MCU version text.

Life0fBrian, 2024-05-31:
I can confirm that this works!
Just install the bd_sensor patch and properly flash the EBB36.

terri1769, 2024-05-31:
No idea what I did differently this time but it appears to be working now.  Thank you.

GromovPeter, 2024-06-08:
hi
mcu 'mcu': Unknown command: I2CBD

root@K1C-5D0B /usr/data/klipper [#] sh ./make_with_bdsensor.sh 
compiling BD_sensor.c into the klipper firmware
make: avr-gcc: Command not found
  Compiling out/src/sched.o
/bin/sh: avr-gcc: not found
make: *** [Makefile:65: out/src/sched.o] Error 127
Updated 1 path from the index
Updated 1 path from the index

what can i do to fix it ?

markniu, 2024-06-09:
seems you have not `make menuconfig` to choose the right mcu type.

> hi mcu 'mcu': Unknown command: I2CBD
>
> root@K1C-5D0B /usr/data/klipper [#] sh ./make_with_bdsensor.sh compiling BD_sensor.c into the klipper firmware make: avr-gcc: Command not found Compiling [...]

GromovPeter, 2024-06-09:
how to know mcu type of k1c ?

McMark1441, 2024-07-13:
I have the same issue. Currently running Octopus in USB Can bridge to my EBB SB2209. I've been working on this for a week with no luck. Flashed, Re-flashed, updated all boards involved.

McMark1441, 2024-07-14:
> you can use the old methond to compile:
> 
> ```
> ~/Bed_Distance_sensor/klipper/install_BDsensor.sh
> make
> ```
> 
> then if the compile and flash goes well there should be a dirty and compile date in the MCU version text.
> 

@terri1769 Please mark this as the solution. 

Seems like you have to run the commands in the following order:

```shell
cd ~/klipper
make clean
make menuconfig 
~/Bed_Distance_ sensor/klipper/install_BDsensor.sh
make
```
Put you tool head into DFU mode, then
```shell
make flash FLASH_DEVICE=XXXX:XXXX
```

terri1769, 2024-07-14:
Resolved

bombsquad85, 2024-08-29:
> > you can use the old methond to compile:
> > ```
> > ~/Bed_Distance_sensor/klipper/install_BDsensor.sh
> > make
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > then if the compile and flash goes well there should be a dirty and compile date in the MCU version text.
> 
> @terri1769 Please mark this as the solution.
> 
> Seems like you have to run the commands in the following order:
> 
> ```shell
> cd ~/klipper
> make clean
> make menuconfig 
> ~/Bed_Distance_ sensor/klipper/install_BDsensor.sh
> make
> ```
> 
> Put you tool head into DFU mode, then
> 
> ```shell
> make flash FLASH_DEVICE=XXXX:XXXX
> ```
> 
> Please correct me if I missed something

This didnt work for me. well, not completely. this got rid of the red MCU error, however when i went to test the bd sensor it said there was an error

bombsquad85, 2024-08-29:
when i follow the directions above, i get rid of the red screen errors, however when i check the connection to the BD sensor i get this: 11:15 PM
No data or corrupt data from BDsensor(), Please check connection
11:15 PM
M102 S-1
11:14 PM
Bed Distance Sensor data error2:10.24

markniu, 2024-08-29:
check the hardware first, which board and pins do you use?

> when i follow the directions above, i get rid of the red screen errors, however when i check the connection to the BD sensor i get this: 11:15 PM No data or corrupt data from BDsensor(��������������������), Please check connection 11:15 PM No data or corrupt data from BDsensor(��������������������), Please check connection 11:15 PM M102 S-1 11:14 PM Bed Distance Sensor data error2:10.24 11:14 PM M119

leftouterjoins, 2024-11-09:
I can confirm the steps as outlined by @McMark1441 solve this issue

terri1769, 2024-11-09:
Resolution here:  https://github.com/markniu/Bed_Distance_sensor/issues/145#issuecomment-2316610804

closed 2024-11-09 by terri1769