markniu/Bed_Distance_sensor #150: Real Time update fail: update z must be >0

Townie-au, 2024-06-11:
BDsensorVer:V1.2c pandapi3d ,switch_mode=1,collision_homing=1,collision_cal=1

About a minute into the print, halts printer.

Townie-au, 2024-06-11:
[
[logs-20240611-154328.zip](https://github.com/user-attachments/files/15782830/logs-20240611-154328.zip)
](url)

markniu, 2024-06-11:
here the actual error is the timer too close, and I will change this message `fail: update z must be >0 ` as a warning message later.
you can add `rt_sample_time:20 ` into the BDsensor section,here 20 means the distance reading sample time is 20ms.
The value of rt_sample_time depends on the MCU speed, it should be higher if the MCU speed is lower.

Townie-au, 2024-06-18:
Different error this time
[Uploading klippy.log (13).txt…]()

markniu, 2024-06-18:
Seems the klippy.log file was not updated successfully.
> Different error this time Uploading klippy.log (13).txt…

Townie-au, 2024-06-18:
CPU: Allwinner H616, Quad-core Cortex-A53 @1.5GHz

markniu, 2024-06-18:
> Different error this time Uploading klippy.log (13).txt…

could you update the klippy.log file again?  and I cann't download this file

Townie-au, 2024-06-18:
[klippy.log (13).txt](https://github.com/user-attachments/files/15884429/klippy.log.13.txt)

markniu, 2024-06-18:
did the error " Unable to read tmc uart 'stepper_z' register GSTAT" happen in the first layer?

Townie-au, 2024-06-18:
Yes

Townie-au, 2024-06-18:
BD sensor and klipper were out of date. 
I updated it this morning. I'll give it another go when I get home.

markniu, 2024-06-19:
> did the error " Unable to read tmc uart 'stepper_z' register GSTAT" happen in the first layer?

and I think this should be the same as the error `timer too close`, you can adjust the the `rt_sample_time: ` to 25 or 30

Townie-au, 2024-06-19:
What does this mean?

Real Time update fail: update z must be >0

3:45 PM

Real time leveling height:0.20 ( 0.20 z_offset:0.00)

Andrew

Townie-au, 2024-06-19:
[klippy.log (14).txt](https://github.com/user-attachments/files/15896216/klippy.log.14.txt)

Townie-au, 2024-06-19:
On the bottom of printer.cfg it has this

747 #*# [BDsensor]

748 #*# z_adjust = 0.0

749 #*# z_offset = 0

Townie-au, 2024-06-19:
So changing rt_sample_time: to 30. Seems to have worked but still getting the real time update fail error in the console but print continuing.

markniu, 2024-06-19:
> What does this mean? Real Time update fail: update z must be >0 3:45 PM Real time leveling height:0.20 ( 0.20 z_offset:0.00) 
that means it will not adjust the z height while printing because the current z position is <0 or higher than setting value

markniu, 2024-06-19:
> > What does this mean? Real Time update fail: update z must be >0 3:45 PM Real time leveling height:0.20 ( 0.20 z_offset:0.00)
> 
> that means it will not adjust the z height while printing because the current z position is <0 or higher than setting value

I have changed this message, and this message is normal sometimes because the z maybe changed in the first layer,for example the intro line in the start gcode in the slicer or retraction

Townie-au, 2024-06-19:
Nope. Shutdown 6% in timer too close EBBCan
[klippy.log (15).txt](https://github.com/user-attachments/files/15899128/klippy.log.15.txt)

markniu, 2024-06-19:
> Nope. Shutdown 6% in timer too close EBBCan [klippy.log (15).txt](https://github.com/user-attachments/files/15899128/klippy.log.15.txt)

did it happen in the first layer? if not make sure there is a `BDSENSOR_SET REAL_TIME_HEIGHT=0` after first layer in the gcode file to disable this feature?

Townie-au, 2024-06-19:
Yeah that's what I thought. I have that gcode in already.
I'll try it with a bed mesh and see how it goes.

feral42, 2024-07-31:
any updates on this?  I'm having the same problem.

Townie-au, 2024-08-01:
I managed to get real time levelling working.

I didnt remove all bed mesh related sections in my printer.cfg and I changed rt_sample_time to 16ms.

closed 2024-08-01 by Townie-au