markniu/Bed_Distance_sensor #151: Error no data

alhanon, 2024-06-11:
after connect sensor to board but sensor information no data this problem after I change main board for printer back to old one same problem no data check wiring but its OK read light working I use BDsensorM

markniu, 2024-06-12:
is there a EXP1 connector on the main board? and recommend to use the EXP1

alhanon, 2024-06-12:
no I use skr pico

markniu, 2024-06-12:
which ports and gpio do you use? some of the ports  on that board can not be used as gpio,for example there are signal filter on the endstop pins that will do filter the communication signal.

markniu, 2024-06-12:
or you can use the gpios on the pi if you use rasspberrypi as the host.

alhanon, 2024-06-12:
I use this gpios before and its work but I remove it to anther printer not work back it to original one not work I think its broken 

[BDsensor] 

sda_pin: gpio22 # example of connecting to main board Creality V4.2.7
scl_pin: gpio29
delay: 20 # 20us per pulse, this value should be >=18 but must be below 30
#z_offset:0 #  within -0.6 to 0.6mm
#z_adjust: -0.02
x_offset: -34
y_offset: 0
no_stop_probe: # fast probe that the toolhead will not stop at the probe point,disable it by commenting out.
position_endstop: 0.5 #the Z axis will be triggered after it measured the distance is below this value(mm) while homing z, recommend value is 0.5~1.0
speed:0.8 # this speed only works for the z tilt and PROBE_ACCURACY command.
collision_homing:0 #  set it 1 to enable homing with nozzle collision sensing.
collision_calibrate:0 # set it 1 to enable auto calibrate BDsensor with nozzle collision sensing.
[force_move] 
enable_force_move: true

markniu, 2024-06-12:
normally, the chip on the sensor will be very hot if it is broken, for example the 5V and GND are invert.
and does the led on the sensor work?

alhanon, 2024-06-12:
It normally not hot . The led work if near bed like normally

alhanon, 2024-06-12:
Flash the firmware normally without problem
[image]

markniu, 2024-06-13:
what's the cable for the BDsensor? I found a few of the old 5 color cables can also cause the connection problem even passed test with multimeter.

alhanon, 2024-06-13:
not cable I change it 3 times .
I lost my hope to work again 

Thank you

closed 2024-06-21 by markniu