markniu/Bed_Distance_sensor #152: QGL Shuts Down Klipper

KiwiChrist, 2024-06-14:
Doing a QGL on my Voron now shuts down klipper firmware on 4th corner when doing first pass.
Worked perfectly until latest klipper update, so I suspect it's to do with Klipper and the QGl command as the error is displayed is "Internal error on command:"_QUAD_GANTRY_LEVEL" ".

Any ideas for a solution?

[image]

markniu, 2024-06-15:
Hello, thanks for feedback this, 
and I just checked that the reason is the official klipper had updated the probe.py today, and the code of BDsensor.py needs to updated too, now I am testing, I will update it to my github late.

KiwiChrist, 2024-06-15:
Yeah... I removed the BDsensor macros and am still getting the error so I cam to the same conclusion. Thanks.

KiwiChrist, 2024-06-15:
Thanks for the quick response and remedy.

closed 2024-06-15 by KiwiChrist