markniu/Bed_Distance_sensor #153: Collision sensing for Marlin

aero4, 2024-06-15:
Hi @markniu!

Because the thermal drift of the probe affects my nozzle to probe offset significantly and I have to adjust this offset nearly every print, the announced "bed collision sensing" would probably eliminate that adjustment and simplify my complex homing routine.

Therefore, my question is, when will collision sensing be available for the Marlin firmware? Are there already some implemented functions for Marlin in the Marlin bugfix branch? When is this feature planned?

Thanks for any information in advance!

markniu, 2024-06-18:
I will add this "bed collision sensing" feature into marlin recently, I think about within one month.
seems the most of the users are from klipper so I spend almost of time to maintain the klipper.

aero4, 2024-06-18:
> I will add this "bed collision sensing" feature into marlin recently, I think about within one month. seems the most of the users are from klipper so I spend most of time to maintain the klipper.

Thanks for the effort, I am looking forward for the implementation!

Will the BD-Sensor M be available soon at AliExpress?

markniu, 2024-07-03:
the collision sensing for the marlin is almost done, https://github.com/markniu/MarlinPULL
I will pull the code to official marlin after more testing.
if someone want to test:
adding `BD_SENSOR_CONTACT_PRBOE` 
increase the speed like: #define Z_PROBE_FEEDRATE_FAST (10*60)
in the configuration.h

aero4, 2024-07-04:
> the collision sensing for the marlin is almost done, https://github.com/markniu/MarlinPULL I will pull the code to official marlin after more testing. if someone want to test: adding `BD_SENSOR_CONTACT_PRBOE` increase the speed like: #define Z_PROBE_FEEDRATE_FAST (10*60) in the configuration.h

Hi @markniu,
thanks alot for your effort! I will try that when I have more spare time.
I am also thinking about switching to Klipper because it has more essential features and runs on my already existing hardware and Marlin was buggy in the last couple of releases...

markniu, 2024-07-11:
Here is a consistency test video about the collision sensing with marlin : https://youtu.be/DzmGyJNS9bY?si=VATehoxtCfTwgFLw

aero4, 2024-07-11:
Just a quick question: do you use collision sensing with a hot or a cold hotend?
If you use a cold one the expansion of the hotend and hardened plastic residues could have some influence?
If you use a hot nozzle wouldn't it burn a hole into the PEI sheet? (the glass transition temperature of PEI is 217°C)

markniu, 2024-07-11:
you are right, the right way is to clean the nozzle and add a heating gcode like M109 S150 before G28.

aero4, 2024-07-11:
> you are right, the right way is to clean the nozzle and add a heating gcode like M109 S150 before G28.

And how about the PEI sheet damage? Does this occur?

markniu, 2024-07-11:
I have never damage a PEI while I develop this feature since beginning of this year that also surprise me.
I think because of the sensitive is depends on the homing z speed not like other pressure sensor.

aero4, 2024-07-11:
> I have never damage a PEI while I develop this feature since beginning of this year that also surprise me. I think because of the sensitive is depends on the homing z speed not like other pressure sensor.

Ok, thanks for the insight! I already damaged some PEI sheets with a hot nozzle but this was always because of an endstop failure (e.g. cloned BLtouch which was unreliable or wrong setup after an upgrade).
How fast (mm/s) do you home the z axis on the first homing attempt?

markniu, 2024-07-11:
the recommend speed of z homing is 3 ~10 mm/s. in the video, both the first and second homing speed are the same 10mm/s .

aero4, 2024-07-11:
> the recommend speed of z homing is 3 ~10 mm/s. in the video, both the first and second homing speed are the same 10mm/s .

Ok, that's my standard homing speed, thanks for the info! So, I can leave the value as it is.

markniu, 2024-07-12:
> > the recommend speed of z homing is 3 ~10 mm/s. in the video, both the first and second homing speed are the same 10mm/s .
> 
> Ok, that's my standard homing speed, thanks for the info! So, I can leave the value as it is.

sorry,  I have verified again that the speed in the video is 5mm/s not 10mm/s.

aero4, 2024-07-12:
> sorry, I have verified again that the speed in the video is 5mm/s not 10mm/s.

Ok, thanks for the info, I will reduce the speed

Carontes, 2024-08-22:
Is it the code already available in the main bugfix branch of Marlin or it has to be added?