markniu/Bed_Distance_sensor #155: Internal error on command:"BED_MESH_CALIBRATE"

legacy23, 2024-06-18:
Hello @markniu 

I am not able to complete Bed Mesh.  Keep getting error "Internal error on command:"BED_MESH_CALIBRATE"" with a "'NoneType' object is not subscriptable" and klipper shutdowns when this happens.   Here is my config and klippy.log

M102 S-1  BDsensorVer:V1.2bpandapi3d.com ,switch_mode=1,collision_homing=1,collision_cal=1

// 1015
// 1015
// 1015
// 1015
// 1015
// 1015
// 1015
// 956
// 956
// 956
// 956
// 949
// 934
// 915
// 892
// 856
// 849
// 825
// 802
// 776
// 751
// 725
// 698
// 673
// 645
// 621
// 593
// 569
// 545
// 520
// 496
// 472
// 449
// 426
// 404
// 382
// 361
// 340
// 320
// 308
$ M102 S-5

[stepper_z]
step_pin: PB6
dir_pin: !PB5
enable_pin: !PC3
microsteps: 16
rotation_distance: 8
#endstop_pin: ^PA7
endstop_pin: probe:z_virtual_endstop
#position_endstop: 0.0
#position_max: 250
position_max: 300
position_min: -8
homing_speed: 5
second_homing_speed: 1  #set this to 3 if homing with collision
homing_retract_speed: 20
homing_retract_dist: 3

[BDsensor] 
sda_pin: PB1
scl_pin: PB0
delay: 10
z_offset: -0.6
z_adjust: -0.3
x_offset: -45
y_offset: -7
#no_stop_probe: 
position_endstop: 2.0 
speed: 0.8
collision_homing: 1 
collision_calibrate: 1

[bed_mesh]
speed: 120
mesh_min: 30,30
mesh_max: 200,210
probe_count: 6,6
horizontal_move_z: 1
algorithm: bicubic
zero_reference_position: 115,115

[klippy.log](https://github.com/user-attachments/files/15880643/klippy.log)

markniu, 2024-06-18:
Hello, you can remove the` #` which is in front of the `no_stop_probe:`

legacy23, 2024-06-19:
Thanks Mark, finally got a my Mesh, I thought “no_stop_probe” was optional.  

I am still having a hard time setting the z_offset,  to get a good first layer I am required to have around 1mm or more and z_offset allows only  0.6mm.

Is the firmware 1.2c required to for it to auto z_offset?

markniu, 2024-06-20:
hello, you can try this: add position_min : 0 in the stepper_z section
[image]

markniu, 2024-06-20:
> Thanks Mark, finally got a my Mesh, I thought “no_stop_probe” was optional.
> 
> I am still having a hard time setting the z_offset, to get a good first layer I am required to have around 1mm or more and z_offset allows only 0.6mm.
> 
> Is the firmware 1.2c required to for it to auto z_offset?

 this “no_stop_probe” is optional now, and I fixed it by sync the code from the latest official klipper.

it should no need to set the z_offset and recommend to keep it to 0 if you have enabled the collision_homing,

closed 2024-06-26 by legacy23