markniu/Bed_Distance_sensor #156: G28 Failure BedMeshCalibrate' object has no attribute '_get_adjusted_points

creek52, 2024-06-20:
I followed the WIKI Install and everything worked until  Step 7. 
I started with a fresh install of Klipper with Fluiid using KIAUH on a Pi3b to talk to a Creality 4.2.7 board. Everything was updated to current . Worked fine with BLTouch probe. Then converted to BD Sensor.
Created new firmware file using './make_with_bdsensor.sh', new bin file was slightly larger than previous bin file. It flashed to the board correctly.
I connected and was able to confirm  sensor is BDsensorVer:V1.2 pandapi3d.com ,switch_mode=0,collision_homing=0,collision_cal=0
I was able to read head distance from bed correctly and calibration data was returned properly.
G28 failed in Step 7.  Attached part of klippy.log and printer.cfg.

I looked at the one issue with a G28 problem and I have no aliases defined any where. That wasn't it.

Thanks.

[BD Sensor V1.2 Problem.txt](https://github.com/user-attachments/files/15917496/BD.Sensor.V1.2.Problem.txt)

markniu, 2024-06-21:
the reason was the official klipper modified the probe code yesterday, and I also sync it into BDsensor.py, please update the BDsensor.py again.

markniu, 2024-06-21:
> the reason was the official klipper modified the probe code yesterday, and I have sync it into BDsensor.py, please update the BDsensor.py again.

creek52, 2024-06-21:
Many Thanks, it has fixed the problem.

closed 2024-06-21 by creek52