markniu/Bed_Distance_sensor #158: INTERNAL_ERROR (klipper) in probe_calibrate, after the first point is read

spaivaras, 2024-06-23:
After recent updates (dont remember which commit was originally, but it was the one that did mech leveling check continues) Cant issue probe_calibrate anymore.  No settings or height was changed

klippy.log
```
Internal error on command:"PROBE_CALIBRATE"
Internal Error on WebRequest: gcode/script
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/webhooks.py", line 256, in _process_request
    func(web_request)
  File "/home/pi/klipper/klippy/webhooks.py", line 436, in _handle_script
    self.gcode.run_script(web_request.get_str('script'))
  File "/home/pi/klipper/klippy/gcode.py", line 229, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 211, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/gcode.py", line 137, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/BDsensor.py", line 458, in cmd_PROBE_CALIBRATE
    self.probe_calibrate_z = curpos[2]
TypeError: 'NoneType' object has no attribute '__getitem__'
```

Console output:
```
11:13 AM Internal error on command:"PROBE_CALIBRATE"
11:13 AM Klipper state: Shutdown
11:13 AM probe at 145.000,130.000 is z=0.000833 (pos:1.097500 - bd:1.097)
11:13 AM probe_calibrate
```

Version info:
```
BDsensor v0.0.0-513-ga24d3e77-inferred
klipper v0.12.0-256-g6d700502
moonraker v0.8.0-379-g346a3d7c
```

spaivaras, 2024-06-23:
It seems its caused by https://github.com/markniu/Bed_Distance_sensor/commit/0c46ce3855cc7bcb08c8fa0f8e74d6ba6a4f0823#diff-bb1ddc1d5ca5bce2021872251f827078db83dc95b2b31a0df31cb7d7bdede2ffL355

removing return statement from `def run_probe`

but `def cmd_PROBE_CALIBRATE(self, gcmd)`  needs the return value:
`curpos = self.run_probe(gcmd)`

I definitely have little knowledge with python, even less with klipper code base, for myself i did a quick hack of adding
`return epos` at the end of `run_probe` function

I see that probe results should be handled differently from probe.results? maybe or maybe gcmd should have set SAMPLES_RESULT == median as run_probe have logic for returning a position? But lacking knowledge here to prepare a merge request

markniu, 2024-06-23:
Thanks！ this bug comes from the sync code with the klipper recently. you can updated the BDsensor.py and try again.

spaivaras, 2024-06-26:
Sorry for the late reply, yep seems probe_calibrate works without issue now, thx for the fix

closed 2024-06-26 by spaivaras