markniu/Bed_Distance_sensor #159: Error 10.24 sfter flashing STLink

rety4, 2024-06-26:
Hello.
I have a problem with BDsensor after flashing STLink. I receive error 10.24. I updated the firmware according to your instructions. Before the upgrade, BDsensor worked ok. Still 'No data from BDsensor,please check connection' on M102 S-1 command.Sensor is connect to EBB36.Should I make any changes after the upgrade?

markniu, 2024-06-26:
does the led on the sensor on and off when there is metal close to the coil?

rety4, 2024-06-26:
Dioda zapala się przy zbliżeniu metalu

markniu, 2024-06-26:
Here is the previous hex version:[stm32autobed_20231102.zip](https://github.com/user-attachments/files/15984547/stm32autobed_20231102.zip)

rety4, 2024-06-26:
Unfortunately the issue still exists. There is no difference which software is uploaded. I programmed both versions using your YT movie as instructions and both times i get notification that software is uploaded correctly. Before first software upgrade sensor worked correctly. What else can i do to fix it?

markniu, 2024-06-27:
could you see the diode flashing, although it flash quickly, but still can see that flashing if it works. 
and it could be better if you can take a video about your process of the uploading.

rety4, 2024-06-27:
I am sending videos of the programming and behavior of the probe. Files were too big, that's why i had to use wetransfer
https://we.tl/t-96ov0m5s4P

markniu, 2024-06-29:
> I am sending videos of the programming and behavior of the probe. Files were too big, that's why i had to use wetransfer https://we.tl/t-96ov0m5s4P

seems that the sensor is working, it should be the connection or cable problem. 
could you try the gpio on the pi for the sda and slc pins?

rety4, 2024-07-01:
I checked the wiring and connection again and the probe still reports an error.
I borrowed a probe from a friend and everything is fine on the second probe.
[image]

markniu, 2024-07-02:
not sure what's the problem, but I think I can send you a new one , please send me(niujl123@sina.com) your shipping address or order number.

rety4, 2024-07-02:
It happens, no problem. I sent the address. Regards Paweł