markniu/Bed_Distance_sensor #160: Center of the probe mark

BittorB, 2024-07-04:
I adapted it to a voron v0 but after calibration(i make several error measuring the offset)  maybe in production is interesting mark the center of the probe phisically for the calibration (you need it to set the offset of the mesh) and to calculate better the offset form nozzle to the probe.

maybe in the case mark a point o a cross in production?

markniu, 2024-07-05:
you can enable the collision sensing and homing z axis 10 times, to see if the output distance value is stable after each homing.

BittorB, 2024-07-07:
the distance can be always the same but can be no centered to the probe or  missaligned from real position to the bed son in this case when you realize the heightmap is missaligned too.
 its only to adjust the X and Y offset to the nozzle in te config.