markniu/Bed_Distance_sensor #163: Internal error during mesh calibrate

heinrichWeichert, 2024-07-13:
Latest HEAD patched into Klipper.

klippy.log attached
[klippy_snip.log](https://github.com/user-attachments/files/16202607/klippy_snip.log)

markniu, 2024-07-14:
do you use the latest klipper? I just tested OK with the latest klipper. the latest code of bdsensor does not work with the old klipper because the official klipper changed a lot about the probe last month.

Townie-au, 2024-07-14:
I had this issue.
horizontal_move_z: in bed_mesh was too high. Changed it to 1.

Andrew

heinrichWeichert, 2024-07-15:
Yes, klipper is up-to-date. I'll check the horizontal_move, thanks.

heinrichWeichert, 2024-07-17:
I changed the values to 
sample_retract_dist: 0.2
And
horizontal_move_z: 0.2

works but there is no second pass 
I guess that has to do with Klipper changes

closed 2024-07-17 by heinrichWeichert