markniu/Bed_Distance_sensor #164: Trouble Adding BD Sensor to Ender 3 Pro with SKR PRO V1.2

Haxim, 2024-07-15:
Struggling to figure out what I'm doing wrong, I feel like I'm so close

Compiling Marlin 2.1.x-bugfix (current as of a few days ago)
Everything is connected correctly (LED on BD Sensor behaves properly when close to metal)
M102 doesn't return as expected (no mention of V1.0 pandapi3d.com):
```
>>> M102 S-1
SENDING:M102 S-1
ok P15 B3
```
M119 behaves oddly. z_min changes from open to TRIGGERED, but z_probe is always set to triggered (I would expect probe to change and z_min to always stay the same)
```
SENDING:M119
Reporting endstop status
x_min: open
y_min: open
z_min: open
z_probe: TRIGGERED
```
M102 S-6 behaves correctly and calibrates, M102 S-5 reads back the calibration curve
Board is a BTT SRK PRO V1.2 (pinouts https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/SKR-PRO-V1.2/manual/SKR-PRO-V1.2PIN.PDF)

I've set the following pins (used to have a BL Touch installed, so the BD Sensor is just using the same connection on the hotend)
```
#ifndef I2C_BD_SDA_PIN
  #define I2C_BD_SDA_PIN                 PA2
#endif

#ifndef I2C_BD_SCL_PIN
  #define I2C_BD_SCL_PIN                 PA1
#endif

#ifndef I2C_BD_DELAY   
  #define I2C_BD_DELAY                   20
#endif

```
Not sure why M102 S-1 isn't responding and z-probe isn't updating correctly. Any ideas?

markniu, 2024-07-15:
for testing, you can then try to  connect to the EXP1, for example  LCD_D7_PG7 and  LCD_D7_PG6 for the SDA and SCL

Haxim, 2024-07-16:
> for testing, you can then try to connect to the EXP1, for example LCD_D7_PG7 and LCD_D7_PG6 for the SDA and SCL

same behaviour (z_min changes state instead of z_probe) on PG7 and PG6 (EXP1 ports suggested for this board). I'm wondering if it's a weird Marlin issue, going to try a different version of the bugfix branch and see if that resolves it.

markniu, 2024-07-16:
did you enable the` USE_PROBE_FOR_Z_HOMING` ?
```
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
// Force the use of the probe for Z-axis homing
#define USE_PROBE_FOR_Z_HOMING
```
and you can this firmware that I pulled to official marlin recently.
https://github.com/markniu/MarlinPULL/tree/bugfix-2.1.x
https://github.com/MarlinFirmware/Marlin/pull/27243

Haxim, 2024-07-19:
Yep, I have using the endstop commented out and use probe for homing
I'll give your fork a shot! Does it also support nozzle collision sensing for Marlin?

markniu, 2024-07-19:
the code on my github supports the collision sensing.

Haxim, 2024-07-19:
Great! I just tried a quick build with your fork and m102 S-1 correctly reports
```
ok P15 B3
BD Sensor version:V1.2c pandapi3d
,sw_mode:1
```
Which seems better than I had before. Now I just need to redo my configurations for this version. Thanks!

Haxim, 2024-07-20:
Works great after setting up my config. Appreciate all the help @markniu

closed 2024-07-20 by Haxim